open62541/open62541 #4125: EUROMAP83 error

zx-zx-zx, 2021-01-17:
## Description
I encountered the following problems when generating euromap83. I found that there is no BSD file about it in nodeset, but I think it is necessary because there is a definition of struct type in the code.
I know the BSD file has been abandoned, but I don't know how to solve this problem.
I need help

## Background Information / Reproduction Steps

C:\Users\ZXT\Desktop\OPC UA\open62541\open62541\tools\nodeset_compiler>python ./nodeset_compiler.py --types-array=UA_TYPES --types-array=UA_TYPES_DI --e ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml  --xml ../../deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml namespace_di
INFO:__main__:Preprocessing (existing) ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing ../../deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed

C:\Users\ZXT\Desktop\OPC UA\open62541\open62541\tools\nodeset_compiler>python ./nodeset_compiler.py --types-array=UA_TYPES --types-array=UA_TYPES_DI --types-array=UA_TYPES_EU83 --e ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --e ../../deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml --xml ../../deps/ua-nodeset/PlasticsRubber/GeneralTypes/1.02/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml namespace_plasticsrubber_generated
INFO:__main__:Preprocessing (existing) ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) ../../deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing ../../deps/ua-nodeset/PlasticsRubber/GeneralTypes/1.02/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
WARNING:nodes:Unknown Field Attribute ArrayDimensions  ------------？？？
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:Windows10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2021-01-20:
If there are any types defined in the nodeset, the bsd should be included in the nodeset.xml (Search for TypeDictionary_BinarySchema or TypeDictionary).

The bsd is base64 encoded and looks like this (as I haven't found the Euromap specifications in the UA-Nodeset repository, I show the MachineTool as a example):
https://github.com/OPCFoundation/UA-Nodeset/blob/9c80cf6e326c8d2f2498d1e5c6efc6218884c97e/MachineTool/Opc.Ua.MachineTool.NodeSet2.xml#L1374-L1447
Just copy the Block of base64 encoded lines to a decoder, e.g. https://emn178.github.io/online-tools/base64_decode.html and save the output to a file.

The bsd-file shoud begin with `<opc:TypeDictionary`, if your file begins with `<xs:schema` you have extracted the base64-encoded xsd file. Have a look for an other base64 encoded block in the nodeset.

If there are no types defined, an empty BSD can be generated:
E.g. replace '${TMP_NSURI}' with the Euromap88 namespace uri in this template.
https://github.com/umati/Sample-Server/blob/693411c0015ff024443a5460b76893dddf9a2d63/cmake/empty.bsd.template#L1-L3

DrHaraldWeber, 2021-01-21:
Created pull request to integrate the official bsd and xsd-files in the OPCFoundation repository
https://github.com/OPCFoundation/UA-Nodeset/pull/77

GoetzGoerisch, 2021-01-27:
The deps/ua-nodeset is updated with #4144

GoetzGoerisch, 2021-02-04:
@zx-zx-zx PR was merged, please verify and update the issue accordingly. Thx

closed 2021-02-04 by zx-zx-zx