open62541/open62541 #4131: How to set custom port number

RavilN, 2021-01-19:
## Description
Cannot find an example of how to set a custom port number in the endpoint URL

## Background Information / Reproduction Steps
If more than one server runs in a single machine, then there is a need to assign different port numbers to each server instance.
In the documentation could not find how to do this simple task.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): - current version.
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

RakshanPremsagar, 2021-01-19:
When you run the server, make use of the following commands
 UA_Server *server = UA_Server_new();
 UA_ServerConfig *config = UA_Server_getConfig(server);
 UA_ServerConfig_setMinimal(config, 4841, NULL);

Here I used 4841 as the port number for my server run. by default it could be 0 and the ports are assigned dynamically if that is  the case.

RavilN, 2021-01-29:
Thank you RakshanPremsagar,
Yes, port number can be set this way, confirmed. 
Cannot find examples of how to set other parameters, like path in the endpoint URL, application name, build info, etc.

RavilN, 2021-01-29:
Actually looking at the function UA_ServerConfig_setMinimalCustomBuffer in file plugins/ua_config_default.c it is possible to figure out, but would be better to have an example where all settings are custom.

teccheck, 2025-06-21:
Currently, adding endpoints in `UA_ServerConfig` overwrites information like endpoint URL with generated values. However, you could modify the endpoints by hand

closed 2025-06-21 by teccheck