open62541/open62541 #4135: Using the stack with an asynchronous socket

epetrevska, 2021-01-21:
## Description

Connecting the open62541 to an asynchronous socket - issues with functions like select, fd_set, fd_isset.

## Background Information / Reproduction Steps

We came across the problem, that the stack needs synchronous socket implementation and we have an asynchronous. Defining a wrapper from synchronous to asynchronous socket is doable, but there are couple of functions like UA_select, UA_fd_set and UA_fd_isset which are incorporated in the open62541 stack, but are used mostly for synchronous I/O calls.

The question is if there is a way to get away with those functions, to deactivate them maybe, or we must implement them somehow? Any help is very much appreciated, thank you.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.4
 - [ ] Other OPC UA SDKs used (client or server):
 - [x] Operating system: for any operating system (or TCP stack) that uses asynchronous sockets
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-01-27:
You are by no means required to use the current network implementation.
People have replaced it by lwip, etc. The architecture-specfic code is under the CC0 license so you can adapt it to local needs without copyleft.

What you describe sounds doable but requires work.
Does every operation have to be asynchronous or can the initialization of the server happen synchronously.
Creating the initial server-socket asynchronously as well is a bit of a pain.

One of our oldest open PR is the network layer overhaul.
https://github.com/open62541/open62541/pull/2408
The brings an event-based network layer that would come close to what you describe.
But this will only happen in a release cycle after 1.2

Did you see the invite to the developer conferene call on the mailing list?
We can also discuss the requirement there.

jpfr, 2021-02-24:
This is being implemented via #4193.

closed 2021-04-15 by NoelGraf