open62541/open62541 #4140: Open62541 not able to connect with Siemens WinCC OPC UA client

MadsHjertstedt-GEA, 2021-01-25:
## Description
Hi guys, 

I'm using the open62541 library for creating an OPC UA server. I've tested the library against many of the common freeware OPC UA clients:
- Prosys Browser
- UA Expert
- Matrikon Explorer
- Integration Object's OPC UA client

With these clients everything works as expected. However when I try to test the server against Siemens WinCC OPC UA client, I'm not able to connect properly and read the address space from the client. 

I've attached a dump from wireshark and the log from open62541. Hope this is enough to find out what is wrong, otherwise let me know what you need.

[image]

[image]

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 
 - [ ] Other OPC UA SDKs used (client or server): WinCC OPC UA client
 - [ ] Operating system: Windows 10 (client on Window Server 2016 Standard)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached 
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2021-01-26:
Can you add a trace with Socket filter, not only opcua.
Is there a FIN,ACK send from client or server side?

MadsHjertstedt-GEA, 2021-01-26:
> Can you add a trace with Socket filter, not only opcua.
> Is there a FIN,ACK send from client or server side?

@mlang-de  you mean like this:
[image]

mlang-de, 2021-01-26:
Yes, if I am right 192.168.079 is the client IP address.
In frame no. 40 the client closes the socket after the server sends the ReadResponse.
So imho this is a problem within the WinCC Client.
(Which node was request in frame no. 38? Is it a structure/extension object or a builtin-variable.)
Anyway, this it no gracefully behavior by the client. Normally a CloseSessionRequest and CloseSecureChannelRequest shall be exchanged before the Socket will be closed.

MadsHjertstedt-GEA, 2021-01-26:
@mlang-de Ok, that doesn't sound good, since my project really depends on the WinCC client. Do you have any idea why other OPC UA servers are able to connect with WinCC then?

Frame no.38 is UA_NS0ID_READREQUEST_ENCODING_DEFAULTBINARY  (631).  This node was also requested in frame no. 35.

mlang-de, 2021-01-26:
NodeId ns=0;i=631 is the request NodeId and not the NodeId to read.
Here you find the requested NodeId.
[image]

MadsHjertstedt-GEA, 2021-01-26:
@mlang-de Oh sorry, I'm still a newbie in the OPC UA community and are still learning, as always.. :-)

The requested NodeIDs are:
[0]: 11705 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERREAD)
[1]: 12165 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERHISTORYREADDATA)
[2]: 12166 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERHISTORYREADEVENTS)
[3]: 11707 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERWRITE)
[4]: 12167 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERHISTORYUPDATEDATA)
[5]: 12168 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERHISTORYUPDATEEVENTS)
[6]: 11709 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERMETHODCALL)
[7]: 11710 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERBROWSE)
[8]: 11711 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERREGISTERNODES)
[9]: 11712 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERTRANSLATEBROWSEPATHSTONODEIDS)
[10]: 11713 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXNODESPERNODEMANAGEMENT)
[11]: 11714 (UA_NS0ID_SERVER_SERVERCAPABILITIES_OPERATIONLIMITS_MAXMONITOREDITEMSPERCALL)

I do **not** have HISTORIZING enabled, so some of the nodes has empty values.

mlang-de, 2021-01-26:
All fine :)
Imho it is ok, that there are NULL or empty values.
UA Spec. Part 5, 1.04 - 6.3.11 defines all childs from OperationLimitsType as optional.
So a client shall be able to handle this.
My assumption is that the WinCC client may can not handle NULL or empty values on the wire?
Which open62541 version you use?

MadsHjertstedt-GEA, 2021-01-26:
@mlang-de Your assumption is perfectly correct. I've changed the value of the 4 nodes related to HISTORIZING to "0" instead of empty value and now the server and client connects and I'm able to browse the AS.

I'm using open62541 version 1.1.1.

mlang-de, 2021-01-26:
Ok nice 👍 
In open62541 v1.2-rc2 are no more NULL Value on the wire (my collegea tells me, not verified :))

MadsHjertstedt-GEA, 2021-01-26:
@mlang-de 
Alright, maybe I'll should check that out. If not to much has changed besides the removal of NULL values, I wouldn't mind upgrading to that version.. 

But anyhow, thank you **so much** for your help troubleshooting this. I was on the edge of loosing it when I discovered the server wouldn't connect with WinCC.

closed 2021-01-26 by MadsHjertstedt-GEA