open62541/open62541 #4141: Please change all Asserts to Fatal Log Messages

rexsum, 2021-01-25:
## Description

Recent changes to the library have caused internal asserts to occur which lead to an immediate termination to an application. Best practices dictate that an internal error, even a critical one, should not terminate with an assert, but should rather log a fatal error message. Please remove all asserts and replace with fatal log messages.

## Background Information / Reproduction Steps

See other posted issues regarding assert errors

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): HEAD
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-01-27:
Asserts are only enabled in Debug builds, not in production builds.
Since you are running a debug build, can you run it in an actual debugger and tell us where the problem occurs?

My guess is, you have mulithreading enabled and use an API method that is not marked as UA_THREADSAFE.
Anyway, we solve these types of issues rather quickly.

rexsum, 2021-01-27:
Julius, you claim that asserts are disabled if I use a Release build, correct? However when I set the cmake build option CMAKE_BUILD_TYPE=Release I see that asserts are still enabled. Please help me understand how to build the library such that the asserts are disabled.

rexsum, 2021-01-27:
That is a good tip to know. I set the cmake option CMAKE_BUILD_TYPE=Release and that disabled the assert error, thx

rexsum, 2021-04-08:
I haven't tested this recently so don't know if it's been changed as originally requested, but the above work-around gets me past the issue.

closed 2021-04-08 by rexsum