open62541/open62541 #4147: Reference loop when creating references manually in specific order

sqadasi, 2021-01-27:
## Description
Hello everyone,
when trying to add "Organizes" references manually to existing Nodes a reference loop can be created, although the added references should not logically build any loops. This behaviour appears when adding a forward "Organizes" reference to a node which has no "Organizes" reference, then using this node as a target for another "Organizes" reference (an inverse reference should be added to that node), but a forward reference is added to this node instead which creates a reference loop (Case 1). 
However, if the reference adding order is changed and the node is first used as a target for "Organizes" reference, then a forward "Organizes" reference is added to it, then it works as it should without any loops (Case 2).

Case 1 (reference loop):
[image]

Case 2:
[image]

Apparently the server fails to distinguish between references with the same Id/Index but different isInverse values for the same node in some cases. I suspected the faulty behaviour was in the check
https://github.com/open62541/open62541/blob/d246ca91d364247e963e0b02e19a289b83b870da/src/server/ua_nodes.c#L568
but changing it to `refs->isInverse != isForward` makes the server not start at all.

## Background Information / Reproduction Steps
I've created a small example to reproduce the problem:

```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>

static void
addNodes(UA_Server *server) {
    UA_NodeId objectNode_1;
	UA_NodeId objectNode_2;
	UA_NodeId objectNode_3;
    UA_NodeId objectNode_4;
	UA_NodeId variableNode_1;
	UA_NodeId variableNode_2;
	UA_NodeId variableNode_3;
    UA_NodeId variableNode_4;
	
    /*
    create the following objects and variables:

    Objects Folder
      |----  Object 1
      |        |----  (HasComponent) Variable 1
      |        |----  (Organizes)    Object 4
      |----  Object 2
      |        |----  (HasComponent) Variable 2
      |----  Object 3
      |        |----  (HasComponent) Variable 3
      |        |----  (HasComponent) Object 4
      |                                |---- (Organizes)    Variable 2
      |                                |---- (Organizes)    Variable 3
      |                                |---- (HasComponent) Variable 4
    */
    UA_ObjectAttributes oAttr = UA_ObjectAttributes_default;
    oAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Object 1");
    UA_Server_addObjectNode(server, UA_NODEID_NULL,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                            UA_QUALIFIEDNAME(1, "Object 1"), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
                            oAttr, NULL, &objectNode_1);
							
    oAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Object 2");
    UA_Server_addObjectNode(server, UA_NODEID_NULL,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                            UA_QUALIFIEDNAME(1, "Object 2"), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
                            oAttr, NULL, &objectNode_2);
							
    oAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Object 3");
    UA_Server_addObjectNode(server, UA_NODEID_NULL,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                            UA_QUALIFIEDNAME(1, "Object 3"), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
                            oAttr, NULL, &objectNode_3);

    oAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Object 4");
    UA_Server_addObjectNode(
        server, UA_NODEID_NULL, objectNode_3,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), UA_QUALIFIEDNAME(1, "Object 3"),
        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE), oAttr, NULL, &objectNode_4);
							

    UA_VariableAttributes variable_1_Attr = UA_VariableAttributes_default;
    UA_String variable_1 = UA_STRING("Some Value 1");
    UA_Variant_setScalar(&variable_1_Attr.value, &variable_1, &UA_TYPES[UA_TYPES_STRING]);
    variable_1_Attr.displayName = UA_LOCALIZEDTEXT("en-US", "Variable 1");
    UA_Server_addVariableNode(server, UA_NODEID_NULL, objectNode_1,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "Variable 1"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), variable_1_Attr, NULL, &variableNode_1);

    UA_VariableAttributes variable_2_Attr = UA_VariableAttributes_default;
    UA_String variable_2 = UA_STRING("Some Value 2");
    UA_Variant_setScalar(&variable_2_Attr.value, &variable_2, &UA_TYPES[UA_TYPES_STRING]);
    variable_2_Attr.displayName = UA_LOCALIZEDTEXT("en-US", "Variable 2");
    UA_Server_addVariableNode(server, UA_NODEID_NULL, objectNode_2,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "Variable 2"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), variable_2_Attr, NULL, &variableNode_2);
							  
    UA_VariableAttributes variable_3_Attr = UA_VariableAttributes_default;
    UA_String variable_3 = UA_STRING("Some Value 3");
    UA_Variant_setScalar(&variable_3_Attr.value, &variable_3, &UA_TYPES[UA_TYPES_STRING]);
    variable_2_Attr.displayName = UA_LOCALIZEDTEXT("en-US", "Variable 3");
    UA_Server_addVariableNode(server, UA_NODEID_NULL, objectNode_3,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "Variable 3"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), variable_3_Attr, NULL, &variableNode_3);
    
    UA_VariableAttributes variable_4_Attr = UA_VariableAttributes_default;
    UA_String variable_4 = UA_STRING("Some Value 4");
    UA_Variant_setScalar(&variable_4_Attr.value, &variable_4, &UA_TYPES[UA_TYPES_STRING]);
    variable_2_Attr.displayName = UA_LOCALIZEDTEXT("en-US", "Variable 4");
    UA_Server_addVariableNode(server, UA_NODEID_NULL, objectNode_4,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "Variable 4"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                              variable_3_Attr, NULL, &variableNode_4);
							  

    /* Adding references */

    UA_ExpandedNodeId target;
	UA_ExpandedNodeId_init(&target);

    /*
    adding an inverse "Organizes reference" first works properly (Case 2)
    */
    /*target.nodeId = objectNode_4;
    UA_Server_addReference(server, objectNode_1,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), target, true);*/

	target.serverIndex = 0;
	target.namespaceUri = UA_STRING_NULL;

    target.nodeId = variableNode_2;
    UA_Server_addReference(server, objectNode_4, UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                           target, true);

    target.nodeId = variableNode_3;
    UA_Server_addReference(server, objectNode_4, UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                           target, true);

    /*
    adding an inverse "Organizes" reference after adding some forward "Organizes" references creates a reference loop (Case 1)
    */
    target.nodeId = objectNode_4;
    UA_Server_addReference(server, objectNode_1, UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                           target, true);

}

/** It follows the main server code, making use of the above definitions. */

static volatile UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    addNodes(server);

    UA_StatusCode retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES:BOOL="1" 
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.2-rc2
 - [X] Other OPC UA SDKs used (client or server): UaExpert (Client)
 - [X] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

juangburgos, 2021-02-04:
Maybe related <https://github.com/open62541/open62541/commit/30a579fd1313713038eb16d568d3ad53b908fdc6#r46753006> ?

sqadasi, 2021-02-05:
@juangburgos indeed it is related. The solution you propused in https://github.com/open62541/open62541/commit/30a579fd1313713038eb16d568d3ad53b908fdc6#r46753006 is also correct. Thank you 👍

jpfr, 2021-02-18:
The proposed solution by @juangburgos was applied to both 1.2 and master.
Closing. Please reopen if an issue remains.

closed 2021-02-18 by jpfr