open62541/open62541 #4151: Build fails on Fedora 34

ctron, 2021-01-28:
## Description

The build fails on Fedora 34 due to:

~~~
/usr/bin/gcc -fPIC -Wno-error=cast-function-type -std=c99 -pipe -Wall -Wextra -Wpedantic -Werror -Wno-static-in-inline -Wno-overlength-strings -Wno-unused-parameter -Wc++-compat -Wformat -Wformat-security -Wformat-nonliteral -Wmissing-prototypes -Wstrict-prototypes -Wredundant-decls -Wuninitialized -Winit-self -Wcast-qual -Wstrict-overflow -Wnested-externs -Wmultichar -Wundef -fno-strict-aliasing -fexceptions -fstack-protector-strong -fstack-clash-protection -fcf-protection -D_FORTIFY_SOURCE=2 -O2 -g -DNDEBUG -flto -fno-fat-lto-objects -Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -shared -Wl,-soname,libopen62541.so.1 -o bin/libopen62541.so.1.1.5 CMakeFiles/open62541-object.dir/open62541.c.o  -lm -lpthread -lrt 
In function 'memcpy',
    inlined from 'UA_String_fromChars' at /builddir/build/BUILD/open62541-1.1.5/build/x86_64-redhat-linux-gnu/open62541.c:7361:9:
/usr/include/bits/string_fortified.h:29:10: error: '__builtin_memcpy' writing one too many bytes into a region of a size that depends on 'strlen' [-Werror=stringop-overflow=]
   29 |   return __builtin___memcpy_chk (__dest, __src, __len,
      |          ^
/builddir/build/BUILD/open62541-1.1.5/build/x86_64-redhat-linux-gnu/open62541.c: In function 'UA_String_fromChars':
/builddir/build/BUILD/open62541-1.1.5/build/x86_64-redhat-linux-gnu/open62541.c:7356:23: note: destination object of size [0, 9223372036854775805] allocated by 'malloc'
 7356 |         s.data = (u8*)UA_malloc(s.length);
      |   
~~~

## Background Information / Reproduction Steps

~~~
-- The C compiler identification is GNU 11.0.0
-- The CXX compiler identification is GNU 11.0.0
~~~

Used CMake options:

```
+ /usr/bin/cmake -S . -B x86_64-redhat-linux-gnu -DCMAKE_C_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_Fortran_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DLIB_SUFFIX=64 -DBUILD_SHARED_LIBS:BOOL=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS=-Wno-error=cast-function-type -DOPEN62541_VERSION=v1.1.5 -DUA_ENABLE_AMALGAMATION=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.12 / 1.1.5
 - [x] Other OPC UA SDKs used (client or server): ??
 - [x] Operating system: Fedora 34
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2021-02-24:
Can't reproduce with gcc10.
The warning in UA_String_fromChars, is there a call stack leading up to the error (similar to clang-anayzer)?

ctron, 2021-02-25:
The only information I have is the one provided. Can you reproduce it with GCC11? Maybe the issue caused by the updated compiler version?

andreasebner, 2025-02-04:
This issue appears to be out of date. If there are known issues, please feel free to open a new issue.