open62541/open62541 #4152: Unable to get user/password authentication to work

tboeckel, 2021-01-28:
First of all, the supplied access control examples are working perfectly. When connecting to the server via UaExpert I am prompted to enter user and password. Although I get a warning requester about "BadSecurityModeInsufficient" I am able to connect and communicate with the example sever.

However, when I add custom users to my own server instance (with exactly the same code as in the examples) I always get an "BadConfigurationError" after entering user and password. Why?

And which instance is generating this error? All occurences of UA_STATUSCODE_BADCONFIGURATIONERROR in the source of open62541 are in PubSub related code, which I am not using at all. So my server instance cannot return this error code. But who is it then? UaExpert? But why when the example program works perfectly?

Used CMake options:
for building the examples
```bash
cmake -DUA_BUILD_EXAMPLES:STRING=ON
```

for building my own code
```bash
cmake -DUA_ENABLE_AMALGAMATION:STRING=ON
```

I build a .so from the amalgamation code to make use of the dynamic library in lots of programs. But I think this should not make any difference.

The relevant code in question from the examples is this:
```c++
static UA_UsernamePasswordLogin logins[2] = {
    {UA_STRING_STATIC("peter"), UA_STRING_STATIC("peter123")},
    {UA_STRING_STATIC("paula"), UA_STRING_STATIC("paula123")}
};

[...]

    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    UA_ServerConfig_setDefault(config);

    /* Disable anonymous logins, enable two user/password logins */
    config->accessControl.clear(&config->accessControl);
    UA_StatusCode retval = UA_AccessControl_default(config, false,
             &config->securityPolicies[config->securityPoliciesSize-1].policyUri, 2, logins);
```

This small excerpt is taken from the server_access_control example, which is working perfectly (apart from the warning). When I paste this code to my code I get the error code mentioned above.

This is the log from UaExpert when connecting to server_access_control:
```
11:17:48.980 | Server Node        | open62541-based OPC UA Appl... | Endpoint: 'opc.tcp://trinity:4840/'
11:17:48.980 | Server Node        | open62541-based OPC UA Appl... | Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#None'
11:17:48.980 | Server Node        | open62541-based OPC UA Appl... | ApplicationUri: 'urn:open62541.server.application'
11:17:48.980 | Server Node        | open62541-based OPC UA Appl... | Used UserTokenType: UserName
11:17:55.331 | Server Node        | open62541-based OPC UA Appl... | The server returned no certificate, all certificate checks will be skipped.
11:19:45.348 | AddressSpaceModel  | open62541-based OPC UA Appl... | Register for ModelChangeEvents failed: createMonitoredItems returned BadNotSupported for item 0
11:19:45.348 | Server Node        | open62541-based OPC UA Appl... | Connection status of server 'open62541-based OPC UA Application' changed to 'Connected'.
```

This is the log of server_access_control itself:
```
[2021-01-28 11:17:55.342 (UTC+0100)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2021-01-28 11:17:55.342 (UTC+0100)] info/channel	Connection 5 | SecureChannel 3 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-01-28 11:17:55.342 (UTC+0100)] info/channel	Connection 5 | SecureChannel 3 | Session 787a1dc1-0bcf-4588-f4ff-bf6b08102122 created
[2021-01-28 11:19:45.343 (UTC+0100)] info/session	SecureChannel 3 | Session ns=1;g=787a1dc1-0bcf-4588-f4ff-bf6b08102122 | ActivateSession: Session activated
[2021-01-28 11:19:45.348 (UTC+0100)] info/session	SecureChannel 3 | Session ns=1;g=787a1dc1-0bcf-4588-f4ff-bf6b08102122 | Subscription 1 | Created the Subscription with a publishing interval of 100.00 ms
[2021-01-28 11:19:45.348 (UTC+0100)] info/session	SecureChannel 3 | Session ns=1;g=787a1dc1-0bcf-4588-f4ff-bf6b08102122 | Subscription 1 | Could not create a MonitoredItem with StatusCode BadNotSupported
[2021-01-28 11:19:45.348 (UTC+0100)] info/session	SecureChannel 3 | Session ns=1;g=787a1dc1-0bcf-4588-f4ff-bf6b08102122 | Subscription 1 | Subscription deleted
```

This is the log from UaExpert when connecting to my own server:
```
11:21:44.065 | Server Node        | savopc localhost               | Endpoint: 'opc.tcp://trinity:51515/'
11:21:44.065 | Server Node        | savopc localhost               | Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#None'
11:21:44.065 | Server Node        | savopc localhost               | ApplicationUri: 'urn:open62541.server.application'
11:21:44.065 | Server Node        | savopc localhost               | Used UserTokenType: UserName
11:21:47.160 | Server Node        | savopc localhost               | The server returned no certificate, all certificate checks will be skipped.
11:21:47.274 | Server Node        | savopc localhost               | Error 'BadConfigurationError' was returned during ActivateSession
11:21:47.286 | Server Node        | savopc localhost               | Connection status of server 'savopc localhost' changed to 'Disconnected'.
```

This is the log of my own server during a failed connection attempt:
```
[2021-01-28 11:21:47.171 (UTC+0100)] info/network	Connection 15 | New connection over TCP from 127.0.0.1
[2021-01-28 11:21:47.171 (UTC+0100)] info/channel	Creating a new SecureChannel
[2021-01-28 11:21:47.171 (UTC+0100)] info/channel	Connection 15 | SecureChannel 2 | Opened SecureChannel
[2021-01-28 11:21:47.171 (UTC+0100)] info/channel	Connection 15 | SecureChannel 2 | Session 5e76a702-de61-a5c5-dd2d-d22d7db7ab3a created
[2021-01-28 11:21:47.174 (UTC+0100)] info/session	Connection 15 | SecureChannel 2 | Session ns=1;g=5e76a702-de61-a5c5-dd2d-d22d7db7ab3a | CloseSession
[2021-01-28 11:21:47.274 (UTC+0100)] warn/channel	Connection 15 | SecureChannel 2 | CloseSession: No Session activated to the SecureChannel
[2021-01-28 11:21:47.275 (UTC+0100)] info/channel	Connection 15 | SecureChannel 2 | CloseSecureChannel
[2021-01-28 11:21:47.275 (UTC+0100)] info/network	Connection 15 | Processing the message failed with error BadConnectionClosed
[2021-01-28 11:21:47.275 (UTC+0100)] info/network	Connection 15 | Closed
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.1.5
 - [X] Other OPC UA SDKs used (client or server): UaExpert 1.5.1
 - [X] Operating system: Linux/Ubuntu 16.04, Linux/Linaro, Linux/Yocto
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tboeckel, 2021-02-04:
I think I found the reason why my own code with user+password was not working as the supplied examples.

I did a small modification in my working copy of open62541, I disabled the two internal user+password entries. This code in plugins/ua_config_default.c
```c++
static const size_t usernamePasswordsSize = 2;
static UA_UsernamePasswordLogin usernamePasswords[2] = {
    {UA_STRING_STATIC("user1"), UA_STRING_STATIC("password")},
    {UA_STRING_STATIC("user2"), UA_STRING_STATIC("password1")}};
```
was replaced by this:
```c++
static const size_t usernamePasswordsSize = 0;
static UA_UsernamePasswordLogin usernamePasswords[0] = {};
```

As it seems the server behaves differently when these zero users are replaced by my own custom users via the UA_AccessControl_default() call from my program instead of when the two default users "user1" and "user2" are replaced.

The big question is: why?

The call of config->accessControl.clear() which defaults to clear_default() should cause no harm. All sizes are checked to be non-zero.

I will leave the original code in place for the moment. But I think the code of open62541 should work without any negative impact even with my tiny modification. I'd really like to get this issue resolved properly.

piperoc, 2023-07-20:
I wonder if anyone saw this. I'm having the same issue but replacing those line makes my compiler go nuts (-pedantic etc. because  of the zero-array).

Is there a cleaner way to do this?

Thanks a lot for posting answers

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier