open62541/open62541 #4156: Build fails when MSYS generator option is enabled

opcua-tsn-team-kalycito, 2021-01-29:
## Description
The objective here is to build the shared libraries of the open62541 stack with MinGW in windows environment. After generating the CMake files, make command fails with the following output.

 
```

Scanning dependencies of target open62541-generator-transport

make[2]: Leaving directory '/c/Users/TESTPC Developer/Documents/test/open62541_latest/open62541/build'

make[2]: Entering directory '/c/Users/TESTPC Developer/Documents/test/open62541_latest/open62541/build'

[ 30%] Generating src_generated/open62541/transport_generated.c, src_generated/open62541/transport_generated.h, src_generated/open62541/transport_generated_handling.h, src_generated/open62541/transport_generated_encoding_binary.h

Traceback (most recent call last):

 File "C:/Users/TESTPC Developer/Documents/test/open62541_latest/open62541/tools/generate_datatypes.py", line 95, in <module>

 parser.create_types()

 File "C:\Users\TESTPC Developer\Documents\test\open62541_latest\open62541\tools/nodeset_compiler/type_parser.py", line 366, in create_types

 self.parse_types()

 File "C:\Users\TESTPC Developer\Documents\test\open62541_latest\open62541\tools/nodeset_compiler/type_parser.py", line 391, in parse_types

 self.parseTypeDefinitions(outname_import, file_import)

 File "C:\Users\TESTPC Developer\Documents\test\open62541_latest\open62541\tools/nodeset_compiler/type_parser.py", line 286, in parseTypeDefinitions

 xmlDoc = etree.iterparse(

 File "C:/dev/msys64/mingw64/lib/python3.8/xml/etree/ElementTree.py", line 1248, in iterparse

 source = open(source, "rb")

OSError: [Errno 22] Invalid argument: 'C;C:\\dev\\msys64\\Users\\TESTPC Developer\\Documents\\test\\open62541_latest\\open62541\\deps\\ua-nodeset\\Schema\\Opc.Ua.Types.bsd'

make[2]: *** [CMakeFiles/open62541-generator-transport.dir/build.make:67: src_generated/open62541/transport_generated.c] Error 1

make[2]: Leaving directory '/c/Users/TESTPC Developer/Documents/test/open62541_latest/open62541/build'

make[1]: *** [CMakeFiles/Makefile2:441: CMakeFiles/open62541-generator-transport.dir/all] Error 2

make[1]: Leaving directory '/c/Users/TESTPC Developer/Documents/test/open62541_latest/open62541/build'

make: *** [Makefile:152: all] Error 2

```

The build was successful when tested with an older SHA : 9917d5a57409a9adc153f96f904a62c4aa7528f3 , and it seems to be broken because of some recent changes, let me know if we have to add something in the CMake flags to get this working.
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -G "MSYS Makefiles" -DBUILD_SHARED_LIBS=ON -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=ON ..
```

jpfr, 2021-09-23:
Is this still current?

kimim, 2021-11-05:
This is the path separator handling issue in Python.
I have a quick fix for this. Wait for a moment.

kimim, 2021-11-05:
This issue is fixed in latest 1.3 branch, could you have a try and close this issue?

teccheck, 2025-06-21:
#4156 is merged, so I'll close this issue

closed 2025-06-21 by teccheck