open62541/open62541 #4163: What to enter for UA_ServerConfig->UA_EndpointDescription->endpointUrl?

JensMertelmeyer, 2021-02-02:
## Description
I have built my first C++ application that creates a OPC UA server with open62541. I have tested it with ["UaExpert" from Unified Automation](https://www.unified-automation.com/products/development-tools/uaexpert.html) and it has been working very well.
I have then tried to use [Softing OPC UA Demo Client](https://data-intelligence.softing.com/products/datafeed-opc-sdks/opc-ua-demo-client/) which also works, but has trouble connecting. It requires an "Endpoint Url" which I apparently do not provide.
[image]

This is the log from open62541:

```
warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
info/network       TCP network layer listening on opc.tcp://My OPC UA Server:4840/
info/network       Connection 300 | New connection over TCP from 192.168.178.65
info/channel       Creating a new SecureChannel
info/channel       Connection 300 | SecureChannel 1 | Opened SecureChannel
info/channel       Connection 300 | SecureChannel 1 | CloseSecureChannel
info/network       Connection 300 | Processing the message failed with error BadConnectionClosed
info/network       Connection 300 | Closed
```
This is the very sparse log from the Softing Demo Client:
```
2021-02-02 13:02:35,477 [4] INFO  - FindServers() returned 1 server(s) from opc.tcp://desktop-g9301hc:4840/ Mask:ClientAPI
2021-02-02 13:02:40,646 [9] ERROR - Invalid URI: The hostname could not be parsed. Mask:ClientAPI

```

## Background Information / Reproduction Steps

I see I that `UA_ServerConfig` has an `UA_EndpointDescription` which has a `UA_STRING endpointUrl`, but I have absolutely no idea what to enter here.

I am creating my OPC server config like this:

```c++
void OPCServer::setServerConfig() {
	auto config = (UA_Server_getConfig(server));
	
	UA_ServerConfig_setCustomHostname(config, UA_STRING_ALLOC("My OPC UA Server"));
	UA_ServerConfig_setDefault(config);
	
	// fill out some things in config->applicationDescription
	
	// fill out some things in config->buildInfo

	config->endpoints->server = config->applicationDescription;
	
	// I have no idea what to enter. How would I know?
	//config->endpoints->endpointUrl = ???
}
```

Used CMake options:
_Using precompiled Win64 from [Release 1.0.2](https://github.com/open62541/open62541/releases/tag/v1.0.2)_

```bash
cmake -???
```

## Checklist

Please provide the following information:

 - [x] open62541 Version: 1.0.2 (70b87b4)
 - [x] Other OPC UA SDKs used: Softing dataFEED OPC UA Client 1.47.0
 - [x] Operating system: Windows 10, 1909
 - [x] Logs attached
 - [ ] Wireshark network dump attached: No
 - [ ] Self-contained code example attached: No
 - [ ] Critical issue: Not yet 😎

mlang-de, 2021-02-02:
What happens if you change
```c
 UA_STRING_ALLOC("My OPC UA Server")
```
to
```c
 UA_STRING_ALLOC("desktop-g9301hc")
```
?

JensMertelmeyer, 2021-02-02:
In that case, the tool will correctly list an endpoint with an url and will happily connect to it:
[image]

Does open62541 offer any help of how to determine the local machine name? In this case, `desktop-g9301hc` is some randomized default value Windows assigned to my virtual machine.

closed 2021-02-02 by JensMertelmeyer