open62541/open62541 #4168: "Serious" server bugs since updated to v1.2

juangburgos, 2021-02-04:
## Description

Since I updated to v1.2 I have found the following issues:

1) I cannot send more than 1 event per emitter in one server cycle. open62541 deletes automatically all events after the first one.

2) References get corrupted if a node has both forward and inverse references of the same reference type.

## Background Information / Reproduction Steps

### Cannot send more than 1 event per emitter 

I don't know exactly what happens, but the relevant source code part is the one marked by the following comment:

* <https://github.com/open62541/open62541/commit/e95a98a542ea94b7a2f19f5019f659f11c857d2c#r46617160>

If there is more than 1 event, the library enters into the remove condition `while(remove > 0)` and deletes all events after the first one without issuing any warning.

### References get corrupted

In this case, I believe the issue is exactly what is marked in the follwing comment:

* <https://github.com/open62541/open62541/commit/30a579fd1313713038eb16d568d3ad53b908fdc6#r46753006>

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [x] Other OPC UA SDKs used (client or server): UaExpert Client
 - [x] Operating system: all
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: N/A
 - [x] Wireshark network dump attached: N/A
 - [x] Self-contained code example attached: explicit code parts shown
 - [x] Critical issue

jpfr, 2021-02-09:
Thanks for making us aware. Will be taken care with priority.

jpfr, 2021-02-18:
Both points were addressed before cutting 1.2.

closed 2021-02-18 by jpfr