open62541/open62541 #4173: Wrong nodeIds when using custom namespace

Sarutobie, 2021-02-09:
## Description
I generated my custom nodeset and can use it in my server.
The problem i am facing right now is that when i use UAExpert to check the server the nodeids of all node are wrong in my custom ObjectNode.

For Example: The nodeset defines a node with ns=3; id=6078. This are generated accordingly. So my namespaces_test.c have the correct nodeids for all nodes. I can see the same nodeid with correct displayname.

In UaExpert i can see all the Nodes, but they have not the same ids as defined in xml. The Node with id 6078 have another id, e.g. 54687.

I am adding the namespace with namespace_test_generated(server);

what am i doing wrong?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL UA_ENABLE_PUBSUB
```

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag): c95fe0f2e0f0433763a23c01958fbdb876e84250
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: ubuntu 20
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Sarutobie, 2021-02-11:
I found out, that all the nodes i actually want to see in my created ObjectNode are available under ObjectTypes->BaseObjectType->EntityType->HardwareEntityType->DeviceType. The last types has to be the custom ones.

So i can see all nodes in the types  but only a part in my ObjectNode.

The Nodes where defines as UAVariables in the model.

I am a bit confused.

Sarutobie, 2021-03-16:
I was reading the OPC specification a lot and the behavior here is correct but if i only now the ids of my types, how do i get the Nodes from the ObjectNode to update the values?

Is there a possibility to look for a specific node with an id?
The examples does not cover this question.

An overview of available functions would be nice.

shaohuazhang1, 2022-08-15:
hello,the id that UaExpert dispayed is sessionId,not nodeid,and the sessionId is automatively.

shaohuazhang1, 2022-08-15:
by the way,I am facing the same question now,if you  have solve this problem,pls share me.thank you!

closed 2026-10-20 by o6-nbeier