open62541/open62541 #4176: [Pub Sub Example] : Error in running the Pub Sub example

Gaurav7611, 2021-02-10:
Hello OPCUA Experts.

I am trying to learn OPCUA Pub Sub. I cloned the publisher [example](https://github.com/open62541/open62541/blob/master/examples/pubsub/tutorial_pubsub_publish.c) from github and tried to run on my linux Ubuntu 20.04 virtual machine.
After cloning the example I compiled and it compiled without any errors.
I tried to run the publisher with the command below :
**./publish opc.udp://224.0.0.22**
After this command, The Publisher is expected to publish the local time every 100 ms but it does not happen.
It continuosly gives error:
**"Sending Network Message Failed"**

**Please find below the command prompt results:**
[2021-02-10 21:15:41.328 (UTC+0100)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-02-10 21:15:41.328 (UTC+0100)] info/server	AccessControl: Anonymous login is enabled
[2021-02-10 21:15:41.328 (UTC+0100)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2021-02-10 21:15:41.328 (UTC+0100)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-02-10 21:15:41.328 (UTC+0100)] info/userland	PubSub channel requested
[2021-02-10 21:15:41.328 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://gaurav-VirtualBox:4840/
[2021-02-10 21:15:41.431 (UTC+0100)] warn/server	PubSub Connection sending failed.
[2021-02-10 21:15:41.431 (UTC+0100)] error/server	PubSub Publish: Could not send a NetworkMessage
[2021-02-10 21:15:41.528 (UTC+0100)] warn/server	PubSub Connection sending failed.
[2021-02-10 21:15:41.528 (UTC+0100)] error/server	PubSub Publish: Could not send a NetworkMessage
[2021-02-10 21:15:41.631 (UTC+0100)] warn/server	PubSub Connection sending failed.
[2021-02-10 21:15:41.631 (UTC+0100)] error/server	PubSub Publish: Could not send a NetworkMessage
[2021-02-10 21:15:41.729 (UTC+0100)] warn/server	PubSub Connection sending failed.
[2021-02-10 21:15:41.729 (UTC+0100)] error/server	PubSub Publish: Could not send a NetworkMessage

--> This mesages are published every 100 ms. and continues to appears until ctrl-c is pressed.
**Kindly guide me regarding how to use the publisher example to see it's functioning**

Any help in this regard is highly appreciated.

Thanks in Advance

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
**cmake -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_PUBSUB=ON ..**
```

## Checklist

Please provide the following information:

 - [ **v1.0.6**] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ Virtual Machine - Ubuntu- 20.04] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2021-02-11:
If I remember correctly, there are parameters that need to be initial used before this example can run.  You can refer to open62541.h on pub sub section for the list of structures and members.

Gaurav7611, 2021-02-11:
> If I remember correctly, there are parameters that need to be initialise before this example can run. You can refer to open62541.h on pub sub section for the list of structures and members.
Thank you for the reply 
Can you please elaborate on this ? Where can I find the information about this ?

Thanks

jackybek, 2021-02-11:
You can take a look at my source code that publishes JSON payload to MQTT broker using dataSetWriter.
The relevant functions for publishing to the broker are from line 1 to line 926 .
From line 927 onwards, they are for subscribing from the broker.
Hope this helps.
`
[SV_PubSub_c.txt](https://github.com/open62541/open62541/files/5967235/SV_PubSub_c.txt)
`

NoelGraf, 2021-04-08:
Has the problem been solved? Can the issue be closed?

closed 2021-04-15 by NoelGraf