open62541/open62541 #4183: Unable to get encryption to work

tboeckel, 2021-02-14:
## Description
I am trying to get basic encryption to work. As a very basic approach I am using the supplied example applications client_encryption and server_encryption. The certificate/key pair has been generated using the create_self-signed.py script and put into /tmp. 

I am running the two applications like this:
```bash
./server_encryption /tmp/server_cert.der /tmp/server_key.der
./client_encryption opc.tcp://localhost:4840 /tmp/server_cert.der /tmp/server_key.der
```

Unfortunately the client is not able to connect to the server.

This is the output of the server:

```
[2021-02-11 06:06:07.562 (UTC+0100)] info/securitypolicy	The Basic128Rsa15 security policy with openssl is added.
[2021-02-11 06:06:07.564 (UTC+0100)] info/securitypolicy	The basic256 security policy with openssl is added.
[2021-02-11 06:06:07.564 (UTC+0100)] info/securitypolicy	The basic256sha256 security policy with openssl is added.
[2021-02-11 06:06:07.564 (UTC+0100)] info/securitypolicy	The Aes128Sha256RsaOaep security policy with openssl is added.
[2021-02-11 06:06:07.564 (UTC+0100)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-02-11 06:06:07.564 (UTC+0100)] info/server	AccessControl: Anonymous login is enabled
[2021-02-11 06:06:07.592 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://trinity:4840/
[2021-02-11 06:06:49.060 (UTC+0100)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2021-02-11 06:06:49.060 (UTC+0100)] info/channel	Connection 5 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-02-11 06:06:49.061 (UTC+0100)] info/channel	Connection 5 | SecureChannel 1 | CloseSecureChannel
[2021-02-11 06:06:49.061 (UTC+0100)] info/network	Connection 5 | Closed
[2021-02-11 06:06:49.061 (UTC+0100)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2021-02-11 06:06:49.062 (UTC+0100)] info/securitypolicy	The Basic128Rsa15 security policy channel with openssl is created.
[2021-02-11 06:06:49.064 (UTC+0100)] info/channel	Connection 5 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15 and a revised lifetime of 600.00s
[2021-02-11 06:06:49.067 (UTC+0100)] warn/channel	Connection 5 | SecureChannel 2 | The client's ApplicationURI did not match the certificate
[2021-02-11 06:06:49.067 (UTC+0100)] info/channel	Connection 5 | SecureChannel 2 | CloseSecureChannel
[2021-02-11 06:06:49.067 (UTC+0100)] info/securitypolicy	The Basic128Rsa15 security policy channel with openssl is deleted.
[2021-02-11 06:06:49.067 (UTC+0100)] info/network	Connection 5 | Closed
^C[2021-02-11 06:07:51.274 (UTC+0100)] info/userland	received ctrl-c
[2021-02-11 06:07:51.274 (UTC+0100)] info/network	Shutting down the TCP network layer
```

This is the output of the client trying to connect to the server:
```
[2021-02-11 06:06:49.058 (UTC+0100)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-02-11 06:06:49.058 (UTC+0100)] info/securitypolicy	The Basic128Rsa15 security policy with openssl is added.
[2021-02-11 06:06:49.059 (UTC+0100)] info/securitypolicy	The basic256 security policy with openssl is added.
[2021-02-11 06:06:49.059 (UTC+0100)] info/securitypolicy	The basic256sha256 security policy with openssl is added.
[2021-02-11 06:06:49.059 (UTC+0100)] info/securitypolicy	The Aes128Sha256RsaOaep security policy with openssl is added.
[2021-02-11 06:06:49.059 (UTC+0100)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2021-02-11 06:06:49.060 (UTC+0100)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2021-02-11 06:06:49.060 (UTC+0100)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2021-02-11 06:06:49.060 (UTC+0100)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2021-02-11 06:06:49.060 (UTC+0100)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2021-02-11 06:06:49.060 (UTC+0100)] info/channel	Connection 3 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-02-11 06:06:49.060 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2021-02-11 06:06:49.060 (UTC+0100)] info/client	Rejecting endpoint 0: security mode doesn't match
[2021-02-11 06:06:49.060 (UTC+0100)] info/client	Rejecting endpoint 1: security mode doesn't match
[2021-02-11 06:06:49.060 (UTC+0100)] info/client	Selected Endpoint opc.tcp://localhost:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2021-02-11 06:06:49.060 (UTC+0100)] info/client	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2021-02-11 06:06:49.061 (UTC+0100)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2021-02-11 06:06:49.061 (UTC+0100)] info/securitypolicy	The Basic128Rsa15 security policy channel with openssl is created.
[2021-02-11 06:06:49.067 (UTC+0100)] info/channel	Connection 3 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15 and a revised lifetime of 600.00s
[2021-02-11 06:06:49.067 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2021-02-11 06:06:49.067 (UTC+0100)] info/client	The ServiceResult has the StatusCode BadSecurityChecksFailed
[2021-02-11 06:06:49.067 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: BadSecurityChecksFailed
[2021-02-11 06:06:49.067 (UTC+0100)] info/securitypolicy	The Basic128Rsa15 security policy channel with openssl is deleted.
[2021-02-11 06:06:49.067 (UTC+0100)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadSecurityChecksFailed
```

I really don't understand why the included examples do not work as expected. What is wrong here?

Used CMake options:
```bash
cmake -DUA_BUILD_EXAMPLES:STRING=ON -DUA_ENABLE_ENCRYPTION_OPENSSL:STRING=ON ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.1.5
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Ubuntu Linux 16.04 (x86_64)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

schroeder-, 2021-02-15:
Looks like the certtool doesn't generate an matching ApplicationURI. The default client uri is "urn:unconfigured:application" and the generate one is by create_self_signed is "urn:open62541.server.application". So they both don't match.

tboeckel, 2021-02-15:
That is a good hint. But I was under the impression that the client has to use the same certificate/key pair as the server. Or does a client require its own (and possibly completely unrelated) certificate/key pair?

tboeckel, 2021-02-15:
Ok, when I create a separate certificate/key pair with ApplicationURI set to "urn:unconfigured:application" I am able to run the client_encryption example and establish a connection to the corresponding server_encryption instance.

The next question is how do I connect to server_encryption with a client like UaExpert? I created another certificate/key pair with ApplicationURI set to "urn:trinity:UnifiedAutomation:UaExpert" as stated in UaExpert's server settings window. First problem is that the generated key has the wrong name extension (.der instead of .pem). This can be easily worked around by either manually entering the key file name or by just changing the name extension from .der to .pem. But the second problem is that UaExpert complains about being unable to load a X509 key:
```
Could not load private key for X509 user token from file '/tmp/uaexpert_key.pem'
```

Does UaExpert require a different certificate/key pair? How do I generate that?

schroeder-, 2021-02-15:
UAExpert can generate certs for it self. Under settings -> manage certs you can generate a new one.

tboeckel, 2021-02-15:
Ok, I created my own certificate+key for UaExpert as client. I set the two files for certificate and key file, set Bsic256Sha256 as security policy and "Sign&Encrypt" as message security mode. As soon as I try to connect to the server_encryption I am prompted to trust the self signed certificate. No problem. But then I get this error log from UaExpert:
```
15:48:46.080 | Server Node        | open62541-based OPC UA Appl... | Endpoint: 'opc.tcp://trinity:4840/'
15:48:46.080 | Server Node        | open62541-based OPC UA Appl... | Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256'
15:48:46.080 | Server Node        | open62541-based OPC UA Appl... | ApplicationUri: 'urn:open62541.server.application'
15:48:46.080 | Server Node        | open62541-based OPC UA Appl... | Used UserTokenType: Certificate
15:48:46.110 | Server Node        | open62541-based OPC UA Appl... | Error 'BadConfigurationError' was returned during ActivateSession
15:48:46.110 | Server Node        | open62541-based OPC UA Appl... | Connection status of server 'open62541-based OPC UA Application' changed to 'Disconnected'.
```

Which part of the configuration is wrong again? Honestly, getting encryption to work is far from being easy or at least straight forward...

schroeder-, 2021-02-15:
I think you mix somethings up. Every client and server has a certificate, which is used for setting up the encrypted traffic. After the server and client established a connection, you have to authenticate to gain access, which is another layer. I think you want to connect as anonymous for authentication method, not via cert.

tboeckel, 2021-02-16:
Yes, you are right. I was under the impression that certificates had to be used for authentication instead of anonymous logins or user+password. The point is that UaExpert allows this kind of authentication, but I failed to see that the client_encryption example still uses the anonymous login.

Many thanks fort your hints!

closed 2021-02-16 by tboeckel