open62541/open62541 #4185: Subsciption callback in c++ opcua client class

linuxsocke, 2021-02-15:
Hi,
## Description
I have written in c++ an client class using the open6241 implementation of opcua. Now I want various instances of this client class. 
Each client instance is connected to a different opcua server. So each instance needs its own subscription callback (no more static callbacks possible then). What is the best way to define this callbacks?

I was thinking using kind of lambda functions. Is this the correct way to go? If yes is can I have a small hint how this works when handing the lambda as c-callback? 

Thanks very much in advance!

ccvca, 2021-02-16:
Lambda-Functions do not work in this case. The only possibility I know is, the node context.
So first set the node context to the this-pointer. Then one common callback use the this-pointer and calls the class function.

linuxsocke, 2021-02-16:
Thanks for the fast answer :) 

So far I mostly learned open62541 just from the Subscriptions_loop.c example (in there is something like *subContext). So I am not familiar with the node context. I can see that there is some clientContext in the ClientConfig:

OpcuaClient::OpcuaClient(){

    client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(cc);
    cc->clientContext = this; 
    cc->stateCallback = tableSubscription;

If I am completly wrong. Is there some example where I can learn more about that?

ccvca, 2021-02-16:
I missed, that you are using an OPC UA client, not a server. In a client, you can set the monContext, which is passed to the callback:

https://github.com/open62541/open62541/blob/0db99235b08359a3bb5844cfc2d02d36eadd70c0/include/open62541/client_subscriptions.h#L179-L185

If all minitored Items of a subscription shall use the same pointer, you can also use the subscriptionContext.

linuxsocke, 2021-02-16:
Ok thanks, so if I understood right somewhere in the initialization it should look like this:

    UA_CreateSubscriptionRequest request = UA_CreateSubscriptionRequest_default();
    UA_CreateSubscriptionResponse response = UA_Client_Subscriptions_create(client_, request,
                                                                            this /*subContext*/, NULL, NULL);
    subId_ = response.subscriptionId;
    UA_MonitoredItemCreateRequest monRequest =
        UA_MonitoredItemCreateRequest_default(UA_NODEID_STRING(1, "the.answer"));
    UA_Client_MonitoredItems_createDataChange(client, subId_,
                                            UA_TIMESTAMPSTORETURN_BOTH,
                                            monRequest, NULL, commonCallback, NULL);
                                            
and in the commonCallback? Its like:

    void commonCallback(
        UA_Client *client, UA_UInt32 subId, void *subContext,
        UA_UInt32 monId, void *monContext, UA_DataValue *value) 
    {
        //how do I call myClassMethod() using subContext
    }

ccvca, 2021-02-16:
```c++
reinterpret_cast<YourClass*>(subContext)->myClassMethod()

linuxsocke, 2021-02-16:
Great it is working. Thanks for your patience :+1:

closed 2021-02-16 by linuxsocke