open62541/open62541 #4190: ABI compatibility policy

traversaro, 2021-02-17:
## Description

Hi open62541 developers, and thanks a lot for the new open62541 release 1.2 and your effort in maintaining and developing open62541. 

I am the maintainer of open62541 in conda-forge (https://conda-forge.org/, https://github.com/conda-forge/open62541-feedstock) and I had a question regarding the ABI compatibility policy of open62541 . 

From the use of soname in Debian packaging (see https://github.com/open62541/open62541/blob/0db99235b08359a3bb5844cfc2d02d36eadd70c0/tools/prepare_packaging.py#L72) I had understood that new minor versions of open62541 were not expected to break ABI compatibility of symbols. However, looking into the new release 1.2 I noticed that there are ABI-breaking changes in some public classes (see for example the changes in `UA_Session` in https://github.com/open62541/open62541/pull/4097).

Could it make sense to document the ABI compatibility policy of open62541 (even if the policy is "we are not looking into ABI, so any version could break it") to avoid that the users make false assumptions?

Thanks a lot in advance.

traversaro, 2021-02-17:
Related issue: https://github.com/open62541/open62541/issues/2923 .

jpfr, 2021-02-18:
We will keep the public API stable for the release families.
That is, the public header files in
- /include or
- /plugins/include
will not be receive breaking changes.

The PR you referenced only changes internal definitions.

That should answer your question.

And kudos for your work on conda-forge!

traversaro, 2021-02-19:
Hi @jpfr, thanks for the answer! 

Just to clarify, I was not asking about [API (Application Programming Interface)](https://en.wikipedia.org/wiki/API), i.e. the fact that the code that compiles fine with open62541 1.x still compiles fine with 1.y (with y > x). Rather, I was asking about [ABI (Application Binary Interface)](https://en.wikipedia.org/wiki/Application_binary_interface), i.e. the about the fact if a program compiled against a shared library  open62541 1.x still works fine without being recompiled with 1.x (with y > x). 

For example, in the PR I linked in https://github.com/open62541/open62541/pull/4097/files#diff-6c538b88fbc0f4639819f585fbb47eb9fd6d5fca12129dec1bfb61eab2a28804L38 the `UA_Session` structure memory layout changed, so the ABI of the library (unless I am missing something) changed. 

To clarify, any kind of ABI policy is fine for me (as I can encode this information easily in conda recipe in https://github.com/conda-forge/open62541-feedstock) so even the policy "we may change ABI at every patch release" is perfectly fine for me, the issue was just about clearly documenting such policy, to avoid confusion for people packaging open62541 in binary-based package managers and that need to deal with such details for assigning `soname`.

jpfr, 2021-02-22:
Ok. For me the ABI was something like x86 with C calling conventions.
We definitely don't have an influence on that.
It seems there are different definitions in use.

With respect to the shared libraries, we will not remove exported symbols within the stable release family. Given that the library was compiled with the same settings.

traversaro, 2021-02-22:
>  It seems there are different definitions in use.

Indeed, sorry for the confusion. For an example of "ABI compatibility" to which I am referring to, se for example the documentation about ABI compatible and incompatible changes of the ABI Laboratory project: https://abi-laboratory.pro/index.php?view=binary-compatibility or of KDE: https://community.kde.org/Policies/Binary_Compatibility_Issues_With_C%2B%2B .

> With respect to the shared libraries, we will not remove exported symbols within the stable release family. 

This is more similar to the definition I was referring to, if by "stable release family" you mean minor releases (i.e. 1.0.* is one family, 1.1.* is another, etc etc) then I would say that you maintain ABI compatibility for each minor release, thanks. Probably this info is relevant for people doing Debian packaging, as the SONAME will need to be changed at every minor release, and not be set equal to the major release number as done in https://github.com/open62541/open62541/blob/0db99235b08359a3bb5844cfc2d02d36eadd70c0/tools/prepare_packaging.py#L72 . 
FYI @Pro @jrcoding

jrcoding, 2021-04-27:
@traversaro @jpfr @Pro 

sorry for my long silence on this issue...

Regarding shared library support the Debian policy (https://www.debian.org/doc/debian-policy/ch-sharedlibs.html) says:

[...]
`The run-time shared library must be placed in a package whose name changes whenever the SONAME of the shared library changes. This allows several versions of the shared library to be installed at the same time, allowing installation of the new version of the shared library without immediately breaking binaries that depend on the old version.`
[...]
`Every time the shared library ABI changes in a way that may break binaries linked against older versions of the shared library, the SONAME of the library and the corresponding name for the binary package containing the runtime shared library should change. Normally, this means the SONAME should change any time an interface is removed from the shared library or the signature of an interface (the number of parameters or the types of parameters that it takes, for example) is changed. This practice is vital to allowing clean upgrades from older versions of the package and clean transitions between the old ABI and new ABI without having to upgrade every affected package simultaneously.`
[...]

So in the Debian world the major release number of the shared lib should change if the ABI changes and the package name should also changes...

As an example, if `v1.0` and `v1.1`of the package named `open62541-1` have the same ABI every thing is fine. If the ABI changes between `v1.0` and `v1.1` then in the Debian world the major number of `v1.1` should change to i.e. `v2.x` and the package name changes to `open62541-2`. The SONAME of the lib then changes respectively...

traversaro, 2021-04-28:
> So in the Debian world the major release number of the shared lib should change if the ABI changes and the package name should also change...

I may be wrong, but it seems to me that there are multiple libraries that break their ABI abd change their SONAME for each new minor release, see for example Boost ( https://packages.debian.org/sid/libboost-filesystem1.74.0, https://abi-laboratory.pro/?view=timeline&l=boost ) or PCL ( https://packages.debian.org/source/sid/pcl ).

jrcoding, 2021-04-28:
> I may be wrong, but it seems to me that there are multiple libraries that break their ABI abd change their SONAME for each new minor release, see for example Boost 

hmmm, yes, you are absolutely right ... don't know what the reason is for this break of the Debian policy in libs like for example Boost... 

Anyway, for me the question still is, do we want and have the chance to follow the Debian policy (changing the major number if ABI changes) or are there objections against the Debian approach?

traversaro, 2021-04-28:
> hmmm, yes, you are absolutely right ... don't know what the reason is for this break of the Debian policy in libs like for example Boost...

I may be wrong, but it seems to me that the Policy says:
> To determine the soversion, look at the SONAME of the library, stored in the ELF SONAME attribute. It is usually of the form name.so.major-version (for example, libz.so.1).

So the soversion = major-version is just something that is "usually" true, not something that is enforced by the policy. 

> Anyway, for me the question still is, do we want and have the chance to follow the Debian policy (changing the major number if ABI changes) or are there objections against the Debian approach?

I guess this is up to the project maintainers, but I want to clarify that this issue is about documenting any ABI policy is adopted in the project, so it would be solved just by documenting the ABI policy wheter it is (even if it "ABI can be broken across minor releases").

jrcoding, 2021-04-28:
not sure that I've got you...?

> I may be wrong, but it seems to me that the Policy says:
>>To determine the soversion, look at the SONAME of the library, stored in the ELF SONAME attribute. It is usually of the form name.so.major-version (for example, libz.so.1).

this describes how to get the `SONAME` of an already installed shared library 
As an example
`objdump -p /usr/lib/x86_64-linux-gnu/libz.so | grep SONAME` gives the SONAME `libz.so.1`
and 
`objdump -p /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.67.0 | grep SONAME` gives the SONAME `libboost_filesystem.so.1.67.0`

So anyway, the SONAME of a library is defined by the maintainers of the library and it would be nice if the SONAME follows some well documented policy so that a user has the chance to detect when the ABI changes. One policy could be the Debian policy but you are right, this decision is up to the project maintainers.
BTW Here is another explanation how commonly the SONAME will be defined (https://doc.dpdk.org/guides/contributing/abi_versioning.html)

As you already correctly noticed, at the moment we have the problem with `libopen62541` as Debian package that the SONAME doesn't strictly change if the ABI changes because at the moment the SONAME is only defined by the major number but the ABI may change if the minor number changes, for example `v1.0` of libopen62541 is not ABI compatible with `v1.1` of libopen62541 but both have the same SONAME `libopen62541.so.1`

So the question to the project maintainers is, how to proceed...???

One solution could be to change the major number of the library when ABI changes (according to the Debian policy).

traversaro, 2021-04-29:
> not sure that I've got you...?

Sorry, my main point was that, as far as I was able to see, there is no point in the Debian policy that enforces that the major version of the library should be bumped when the SONAME changes.

jrcoding, 2021-04-29:
> > not sure that I've got you...?
> 
> Sorry, my main point was that, as far as I was able to see, there is no point in the Debian policy that enforces that the major version of the library should be bumped when the SONAME changes.

ahhh, sorry, now I got you, of course you are right, it's only just a common way to bump the major version if the ABI changes but of course it's not a must...

But the SONAME of the shared library should definitely change if the ABI changes (FYI see also the discussion: https://github.com/open62541/open62541/pull/2367#issuecomment-454370297) 

So the question to upstream is how to detect in a save way that the ABI of a new version of open62541 changes (major number?, minor number?, combination of both?) so that one can build shared library packages with different SONAMES automatically in conjunction with an ABI change?

guillemj, 2024-09-29:
Hi!

To add to what has been said already. The SONAME should match the library ABI, and it does not necessarily need to match the API, some projects manage two independent numbers for these. If the project wants to go with semver (as proposed in #2923) , then that might tie both of these numbers.

If the project is currently not ABI stable and there are plans to break ABI on each minor release, then that should indeed be represented by bumping the SONAME on each of these occasions. Take into account thought that for a downstream like Debian (but I'd say this applies to any downstream really), this implies a library transition. In the Debian context that means, requesting a transition slot with the release team depending on the amount of reverse dependencies, renaming the binary package for the library which requires manual approval from the archive admins (and that might take an undetermined amount of time), and then the release team triggering an automatic rebuild of all these reverse dependencies. If the API also changed, then that would in addition imply preparing patches for all the reverse dependencies that would fail to build, and that would require manually applying those patches and manually uploading these packages, which can be tedious work. (Describing this so that you are aware of the implications).

For some projects that have both a very unstable API and ABI, they could be packaged with one source package by incompatible release (for example boost), but that is in general very much frowned upon, as that duplicates packages, and can be even more work to maintain and handle, so I don't think this is generally a viable option, unless there are very exceptional circumstances.