open62541/open62541 #4194: Integeroverflows in atoi functions

dns43, 2021-02-20:
## Description

Easy to fix uint_64 overflows in atoi functions via very large input.

## Background Information / Reproduction Steps
atoiSigned
https://github.com/open62541/open62541/blob/bb8ce4be345b23ab84606e91f05be06d446cfb6e/deps/atoi.c#L66

atoiUnsigned
https://github.com/open62541/open62541/blob/bb8ce4be345b23ab84606e91f05be06d446cfb6e/deps/atoi.c#L37

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): bb8ce4be345b23ab84606e91f05be06d446cfb6e
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:  Ubunbtu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-02-22:
That's right. Imho we should not maintain an internal atoi implementation.
Pull Requests to move to a standard libc implementation are welcome.

andreasebner, 2025-07-30:
The atoi implementation is not in the repo anymore.

closed 2025-07-30 by andreasebner