open62541/open62541 #4197: "OPENSSL_API_COMPAT" is not defined

makogo82, 2021-02-22:
**I am trying to compile open62541 on redhat 7. with the current openssl version
OpenSSL 1.0.2k-fips  26 Jan 2017**
I am using the following cmake command: 
 cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL  ..

/tmp/open62541/plugins/crypto/openssl/securitypolicy_openssl_common.c: In function 'UA_Openssl_Init':
/tmp/open62541/plugins/crypto/openssl/securitypolicy_openssl_common.c:72:6: error: "OPENSSL_API_COMPAT" is not defined [-Werror=undef]
 #if  OPENSSL_API_COMPAT < 0x10100000L

When I tried with: 
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DOPENSSL_API_COMPAT=1.02 ..
I got a lot of errors about the certiifcates.

DO I need to use some specific git branch when I am using OpenSSL 1.0.2k

I was able to compile ad execute open62541  OpenSSL 1.1 installed on fedora

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-02-24:
Fixed via #4200.
Will be released with the next v1.2 patch release.

closed 2021-02-24 by jpfr

makogo82, 2021-02-26:
The issue is fixed for plugins/crypto/openssl/securitypolicy_openssl_common.c
but now I got more errors (I am using gcc version 5.3.1 20160406 (GCC)):

/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c: In function â��UA_CertificateVerification_Verifyâ��:
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:481:5: error: "OPENSSL_API_COMPAT" is not defined [-Werror=undef]
#if OPENSSL_API_COMPAT < 0x10100000L
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:509:20: error: implicit declaration of function â��X509_get_key_usageâ�� [-Werror=implicit-function-declaration]
uint32_t val = X509_get_key_usage(certificateX509);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:509:5: error: nested extern declaration of â��X509_get_key_usageâ�� [-Werror=nested-externs]
uint32_t val = X509_get_key_usage(certificateX509);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:509:20: error: conversion to â��uint32_t {aka unsigned int}â�� from â��intâ�� may change the sign of the result [-Werror=sign-conversion]
uint32_t val = X509_get_key_usage(certificateX509);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:572:38: error: implicit declaration of function â��X509_get0_subject_key_idâ�� [-Werror=implicit-function-declaration]
trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:572:17: error: nested extern declaration of â��X509_get0_subject_key_idâ�� [-Werror=nested-externs]
trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:572:36: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:575:35: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
remote_cert_keyid = X509_get0_subject_key_id(certificateX509);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c: At top level:
cc1: error: unrecognized command line option â��-Wno-static-in-inlineâ�� [-Werror]
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/ua_pki_openssl.c.o] Error 1
make[1]: *** [CMakeFiles/open62541-plugins.dir/all] Error 2
make: *** [all] Error 2

makogo82, 2021-02-26:
Can you also performed some compilation tests with openssl1.0.2