open62541/open62541 #4199: UA_FieldMetaData initialisation for char(255) type

jackybek, 2021-02-24:
## Description
According to the header definition, this is the UA_FieldMetaData definition.  In the example codes under pubsub directory, only int16, int32, date time and float data type are illustrated.
There are no String data types in the example.

typedef struct {
    UA_String name;
    UA_LocalizedText description;
    UA_DataSetFieldFlags fieldFlags;
    UA_Byte builtInType;
    UA_NodeId dataType;
    UA_Int32 valueRank;
    size_t arrayDimensionsSize;
    UA_UInt32 *arrayDimensions;
    UA_UInt32 maxStringLength;
    UA_Guid dataSetFieldId;
    size_t propertiesSize;
    UA_KeyValuePair *properties;
} UA_FieldMetaData;

I realised that in order to prepare the meta data definition for string type, it requires additional parameters to be set (as compared to float type)..

For example, if I did not set maxStringLength, arrayDimensions, and arrayDimensionsSize, I will get segmentation fault.  

question: for a variable of type char myString[255], what should I set for the parameter : arrayDimensions 

see below for more details on my attempted settings in my code.
What other fields needs to be initialized?

## Background Information / Reproduction Steps
======================================
UA_FieldMetaData_init (&pMetaData->fields[n]);
pMetaData->fields[n].builtInType = UA_NS0ID_STRING;
UA_NodeId_copy (&UA_TYPES[UA_TYPES_STRING].typeId, &pMetaData->fields[n].dataType);
pMetaData->fields[n].valueRank = 1; // 1 dimension;
pMetaData->fields[n].arrayDimensionsSize = 1;
pMetaData->fields[n].maxStringLength = 255;
pMetaData->fields[n].name = UA_STRING("2. Block Version");

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag): v2.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: Linux 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ x] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2021-02-24:
I am referring to this example:
open62541/examples/pubsub/tutorial_pubsub_subscribe.c
As you can see there are no metafield of type String.

opcua-tsn-team-kalycito, 2021-02-25:
Hi Jacky,

I am not sure why you are using char[255] instead of String type itself.

You can create a variable node of type String in the publisher side, copy your string into it and publish it. On the subscriber side, if you have setup the metadata configuration correctly to receive the string it should work properly. We have verified it at our end by modifying the tutorial_pubsub_publish and tutorial_pubsub_subscribe accordingly and it works fine.

jackybek, 2021-02-25:
https://github.com/open62541/open62541/blob/master/examples/pubsub/tutorial_pubsub_subscribe.c

Line 230-270 does not contain any metadata field of type string.  Can you point me to the correct file ?

jackybek, 2021-02-25:
You mentioned : On the subscriber side, if you have setup the metadata configuration correctly to receive the string it should work properly.’
That’s exactly what I would like to know how to setup the metadata configuration for a string type.  On your examples in GitHub, there isn’t any.  Can you show me the proper way to setup metadata configuration for string type ?

jackybek, 2021-03-01:
Thanks.
Based on your hint, i manage to solve part 1 of the problem.
Thanks.

closed 2021-03-01 by jackybek