open62541/open62541 #4201: High jitter for async method calls

dapaulid, 2021-02-24:
In our tests with asynchronous server methods, we observed an average jitter of 50 ms from client point of view. The reason for this seems to be that sending of the response is delayed by a 100 ms timer:

https://github.com/open62541/open62541/blob/5cc2be0137f674dd2da4dee362d1f03d8137f939/src/server/ua_server_async.c#L162-L165

https://github.com/open62541/open62541/blob/5cc2be0137f674dd2da4dee362d1f03d8137f939/src/server/ua_server_async.c#L107-L151

As a workaround, we tried to reduce the timer interval to 1 ms, but this introduces additional CPU load due to excessive context switching (server probably spends less time in the `select()` call).

A solution could be to call `processAsyncResults` directly from `UA_Server_setAsyncOperationResult`, i.e. in the context of the application thread. However, I'm not sure what this would mean concerning thread safety, since a comment states that `processAsyncResults` is only executed by the server thread.

What do you think?

jpfr, 2021-02-24:
Can you explain the setup in more detail?
Do you spin up a worker thread to process the async operations?

dapaulid, 2021-02-24:
In our setup, we have an open62541 client that periodically calls a method on a open62541 server. Using synchronous method calls, the call took approx. 1 ms to return a result.
 
We now switched to asynchronous calls to ensure that the server is always responsive, i.e. we use `UA_Server_setAsyncOperationResult` from a worker thread. However, we found that the same calls now may take up to 100 ms due to the delayed sending of the response. This makes it barely usable for our case, as we need to process calls with high frequency.