open62541/open62541 #4204: Customize user permissions - Hide nodes

RFil, 2021-02-25:
When a user has no rights to read a node, i want to completely hide that node to the client.
What I have got at the moment is:
- customizing the callback myConfig->accessControl.allowBrowseNode => The node gets visible but not browsable (no contents in it)
- customizing the callback myConfig->accessControl.getUserAccessLevel => The 'value' attribute of a property gets visible/not-visible or writable/not-writable

I'm digging into the use of the callback myConfig->accessControl.getUserRightsMask
It gets called, but the returned result seems to have no effect. I see no difference between returning 0x00000000u and returning 0xffffffffu.

Moreover, I don't 100% understand the meanings of the UA_WRITEMASK_xxx (Write Masks in the documentation). They shuld be the (OR-ed) return value of myConfig->accessControl.getUserRightsMask
In the open62541 code, they looks used in macro CHECK_USERWRITEMASK(mask), in turn used in copyAttributeIntoNode 

//#define UA_WRITEMASK_ACCESSLEVEL             (0x01u << 0u)
//#define UA_WRITEMASK_ARRRAYDIMENSIONS        (0x01u << 1u)
//#define UA_WRITEMASK_BROWSENAME              (0x01u << 2u)
//#define UA_WRITEMASK_CONTAINSNOLOOPS         (0x01u << 3u)
//#define UA_WRITEMASK_DATATYPE                (0x01u << 4u)
//#define UA_WRITEMASK_DESCRIPTION             (0x01u << 5u)
//#define UA_WRITEMASK_DISPLAYNAME             (0x01u << 6u)
//#define UA_WRITEMASK_EVENTNOTIFIER           (0x01u << 7u)
//#define UA_WRITEMASK_EXECUTABLE              (0x01u << 8u)
//#define UA_WRITEMASK_HISTORIZING             (0x01u << 9u)
//#define UA_WRITEMASK_INVERSENAME             (0x01u << 10u)
//#define UA_WRITEMASK_ISABSTRACT              (0x01u << 11u)
//#define UA_WRITEMASK_MINIMUMSAMPLINGINTERVAL (0x01u << 12u)
//#define UA_WRITEMASK_NODECLASS               (0x01u << 13u)
//#define UA_WRITEMASK_NODEID                  (0x01u << 14u)
//#define UA_WRITEMASK_SYMMETRIC               (0x01u << 15u)
//#define UA_WRITEMASK_USERACCESSLEVEL         (0x01u << 16u)
//#define UA_WRITEMASK_USEREXECUTABLE          (0x01u << 17u)
//#define UA_WRITEMASK_USERWRITEMASK           (0x01u << 18u)
//#define UA_WRITEMASK_VALUERANK               (0x01u << 19u)
//#define UA_WRITEMASK_WRITEMASK               (0x01u << 20u)
//#define UA_WRITEMASK_VALUEFORVARIABLETYPE    (0x01u << 21u)

Thanks all in advance

 - [ 1.1.5] open62541 Version (release number or git tag):
 - [ UaExpert] Other OPC UA SDKs used (client or server): 
 - [Win 10 - Ubuntu 20 ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

dns13, 2023-09-20:
Have you found a way to do so? I want to do the same but haven't found a way either.

The only possible way I found by now is to stuff all nodes I want to "hide" into a specific node where I can disallow browsing for, but thats kind of ugly and does not really fit my needs. Also the top level node is still visible to all users.

RFil, 2023-09-20:
> Have you found a way to do so? I want to do the same but haven't found a way either.
>
> The only possible way I found by now is to stuff all nodes I want to "hide" into a specific node where I can [...]

I confirm that AllowBrowse allows the node to browse his childrens, not the node himself. The node remains still visible because HIS parent has AllowBrowse. I've come to terms with it
Sorry, no better solution

dns13, 2023-09-21:
Thank you anyways!

FeiSuNOV, 2024-09-12:
Is it possible to use allowBrowseNode to hide some nodes, but make other nodes visible.
I tried to use allowBrowseNode, but it hide all nodes from root level.

FeiSuNOV, 2024-09-12:
just add some logic to check the node id, and return true / false depend on the id, seems working.

aadiljaleel, 2025-01-23:
Any update on this?
And did anyone figure what's up with getUserRightsMask()? I also get no change returning 0x00.. or 0xFF.. from it.