open62541/open62541 #4208: BadUserAccessDenied by Logging with User0/User1

MichaelKlin, 2021-02-26:
## Description
i am running a server on Debian9.
i added some users according to the example for the server_access_control.
I give each user different permissions(loot at the code). IF i log with User Login and Password, i get in UAexpert BadUserAccessDenied. And every 10 sec server is disconnected and connect

 
## Background Information / Reproduction Steps

**My AccessControl.c**

```
static UA_Byte myGetUserAccesRights(UA_Server* server, UA_AccessControl* ac, const UA_NodeId* sessionId,
	void* sessionContext, const UA_NodeId* nodeId, void* nodeContext)
{
	/*printf("User Acces Rights\n");*/
	UA_Byte res = 0;
	if (*(int*)sessionContext == User1RIGHTS) {
		res = UA_ACCESSLEVELMASK_READ;
	}
	else if (*(int*)sessionContext == User2RIGHTS)
		res = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

	return UA_FALSE;
}
```

```
UA_StatusCode retval=UA_AccessControl_default(config, false,
		&config->securityPolicies[config->securityPoliciesSize - 1].policyUri, 2, logins); 
config->accessControl.getUserAccessLevel = myGetUserAccesRights;

```

**And i changed  activateSession_default**

```

        UA_Boolean match = false;
        for(size_t i = 0; i < context->usernamePasswordLoginSize; i++) {
            if(UA_String_equal(&userToken->userName, &context->usernamePasswordLogin[i].username) &&
               UA_String_equal(&userToken->password, &context->usernamePasswordLogin[i].password)) {
                match = true;
                *(size_t*)sessionContext = i + 1;
                break;
            }
        }
        if(!match)
            return UA_STATUSCODE_BADUSERACCESSDENIED;

```
Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1.2
 - [x] Other OPC UA SDKs used (client or server): UAexpert 
 - [x] Operating system: Debian9
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

MichaelKlin, 2021-02-26:
[image]

MichaelKlin, 2021-03-04:
Solved

closed 2021-03-04 by MichaelKlin

piperoc, 2023-07-20:
@EvgKlin  care to share how you solved this?