open62541/open62541 #4211: "OPENSSL_API_COMPAT" is not defined in openssl_common.c

makogo82, 2021-02-28:
## Description

This issue is related to #4197 

but now I got more errors (I am using gcc version 5.3.1 20160406 (GCC)):

/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c: In function â��UA_CertificateVerification_Verifyâ��:
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:481:5: error: "OPENSSL_API_COMPAT" is not defined [-Werror=undef]
#if OPENSSL_API_COMPAT < 0x10100000L
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:509:20: error: implicit declaration of function â��X509_get_key_usageâ�� [-Werror=implicit-function-declaration]
uint32_t val = X509_get_key_usage(certificateX509);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:509:5: error: nested extern declaration of â��X509_get_key_usageâ�� [-Werror=nested-externs]
uint32_t val = X509_get_key_usage(certificateX509);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:509:20: error: conversion to â��uint32_t {aka unsigned int}â�� from â��intâ�� may change the sign of the result [-Werror=sign-conversion]
uint32_t val = X509_get_key_usage(certificateX509);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:572:38: error: implicit declaration of function â��X509_get0_subject_key_idâ�� [-Werror=implicit-function-declaration]
trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:572:17: error: nested extern declaration of â��X509_get0_subject_key_idâ�� [-Werror=nested-externs]
trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:572:36: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:575:35: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
remote_cert_keyid = X509_get0_subject_key_id(certificateX509);
^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c: At top level:
cc1: error: unrecognized command line option â��-Wno-static-in-inlineâ�� [-Werror]
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/ua_pki_openssl.c.o] Error 1
make[1]: *** [CMakeFiles/open62541-plugins.dir/all] Error 2
make: *** [all] Error 2
Used CMake options:

## Background Information / Reproduction Steps

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DOPENSSL_API_COMPAT=1.0.2
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tboeckel, 2021-03-01:
Even with the changes introduced by #4200 I am still not able to build open62541. I no longer get errors about OPENSSL_API_COMPAT being undefined, but functions like X509_get_key_usage() and X509_get0_subject_key_id() are still called. These do not yet exist in OpenSSL 1.0.2g, which I am using on Ubuntu 16.04 (yes, I know, it is outdated...)

smitha-cgi, 2021-03-01:
@tboeckel If you revert [this commit](https://github.com/open62541/open62541/commit/72af2b1efdcd219e62aed060c39eeaada5468244) you'll be able to build against 1.0.2g

makogo82, 2021-03-01:
> @tboeckel If you revert [this commit](https://github.com/open62541/open62541/commit/72af2b1efdcd219e62aed060c39eeaada5468244) you'll be able to build against 1.0.2g

Now I got those errors (gcc version 5.3.1 20160406):

/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c: In function ‘UA_CertificateVerification_Verify’:
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:505:20: error: implicit declaration of function ‘X509_get_key_usage’ [-Werror=implicit-function-declaration]
     uint32_t val = X509_get_key_usage(certificateX509);
                    ^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:505:5: error: nested extern declaration of ‘X509_get_key_usage’ [-Werror=nested-externs]
     uint32_t val = X509_get_key_usage(certificateX509);
     ^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:505:20: error: conversion to ‘uint32_t {aka unsigned int}’ from ‘int’ may change the sign of the result [-Werror=sign-conversion]
     uint32_t val = X509_get_key_usage(certificateX509);
                    ^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:568:38: error: implicit declaration of function ‘X509_get0_subject_key_id’ [-Werror=implicit-function-declaration]
                 trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
                                      ^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:568:17: error: nested extern declaration of ‘X509_get0_subject_key_id’ [-Werror=nested-externs]
                 trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
                 ^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:568:36: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
                 trusted_cert_keyid = X509_get0_subject_key_id(trusted_cert);
                                    ^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c:571:35: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
                 remote_cert_keyid = X509_get0_subject_key_id(certificateX509);
                                   ^
/tmp/open62541/plugins/crypto/openssl/ua_pki_openssl.c: At top level:
cc1: error: unrecognized command line option ‘-Wno-static-in-inline’ [-Werror]
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/ua_pki_openssl.c.o] Error 1
make[1]: *** [CMakeFiles/open62541-plugins.dir/all] Error 2
make: *** [all] Error 2

mpfj, 2021-03-01:
Same for me ... I just fixed it with the following mini-patch:-

```
diff --git a/plugins/crypto/openssl/ua_pki_openssl.c b/plugins/crypto/openssl/ua_pki_openssl.c
index ea18796..d66f34e 100644
--- a/plugins/crypto/openssl/ua_pki_openssl.c
+++ b/plugins/crypto/openssl/ua_pki_openssl.c
@@ -502,6 +502,7 @@ UA_CertificateVerification_Verify (void *                verificationContext,
     }
 #endif
 
+#if OPENSSL_VERSION_NUMBER >= 0x1010000fL
     /* This condition will check whether the certificate is a User certificate or a CA certificate.
      * If the KU_KEY_CERT_SIGN and KU_CRL_SIGN of key_usage are set, then the certificate shall be
      * condidered as CA Certificate and cannot be used to establish a connection. Refer the test case
@@ -511,6 +512,7 @@ UA_CertificateVerification_Verify (void *                verificationContext,
        (val & KU_CRL_SIGN)) {
         return UA_STATUSCODE_BADCERTIFICATEUSENOTALLOWED;
     }
+#endif
 
     opensslRet = X509_verify_cert (storeCtx);
     if (opensslRet == 1) {
@@ -557,6 +559,7 @@ UA_CertificateVerification_Verify (void *                verificationContext,
     else {
         opensslRet = X509_STORE_CTX_get_error (storeCtx);
 
+#if OPENSSL_VERSION_NUMBER >= 0x1010000fL
         /* Check the issued certificate of a CA that is not trusted but available */
         if(opensslRet == X509_V_ERR_SELF_SIGNED_CERT_IN_CHAIN){
             int                     trusted_cert_len = sk_X509_num(ctx->skTrusted);
@@ -582,6 +585,7 @@ UA_CertificateVerification_Verify (void *                verificationContext,
                 }
             }
         }
+#endif
 
         /* Return expected OPCUA error code */
         ret = UA_X509_Store_CTX_Error_To_UAError (opensslRet);
```

For my testing, I can still connect using my self-cert certificate using openssl 1.0.1e installed on our target hardware.

makogo82, 2021-03-01:
@mpfj nice job.

thegitworker, 2021-06-07:
What was done to solve this? I am getting the same error from the OP. I applied the patch and still receive the same error. Did you revert the commit as well?

Here's what I did:
````bash
git submodule update --init --recursive
saved above patch in mini.patch
patch -i mini.patch
mkdir build && cd build
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION_OPENSSL=ON ..
make
...
Scanning dependencies of target open62541-plugins
[  7%] Building C object CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/securitypolicy_openssl_common.c.o
[  9%] Building C object CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/ua_openssl_basic128rsa15.c.o
[  9%] Building C object CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/ua_openssl_basic256.c.o
[ 11%] Building C object CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/ua_openssl_basic256sha256.c.o
[ 12%] Building C object CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/ua_openssl_aes128sha256rsaoaep.c.o
[ 12%] Building C object CMakeFiles/open62541-plugins.dir/plugins/crypto/openssl/ua_pki_openssl.c.o
/home/viewr/open62541/plugins/crypto/openssl/ua_pki_openssl.c:481:5: error: 'OPENSSL_API_COMPAT' is not defined, evaluates to 0
      [-Werror,-Wundef]
#if OPENSSL_API_COMPAT < 0x10100000L
    ^
1 error generated.
*** Error code 1

Stop.
make[2]: stopped in /usr/home/viewr/open62541/build
*** Error code 1

Stop.
make[1]: stopped in /usr/home/viewr/open62541/build
*** Error code 1

Stop.
make: stopped in /usr/home/viewr/open62541/build
````
If I revert the commit from above the patch seems to fail, whereas the above didn't fail any.
````bash
patch -i mini.patch 
Hmm...  Looks like a unified diff to me...
The text leading up to this was:
--------------------------
|diff --git a/plugins/crypto/openssl/ua_pki_openssl.c b/plugins/crypto/openssl/ua_pki_openssl.c
|index ea18796..d66f34e 100644
|--- a/plugins/crypto/openssl/ua_pki_openssl.c
|+++ b/plugins/crypto/openssl/ua_pki_openssl.c
--------------------------
Patching file plugins/crypto/openssl/ua_pki_openssl.c using Plan A...
Hunk #1 failed at 502.
Hunk #2 succeeded at 485 with fuzz 2 (offset -27 lines).
Hunk #3 failed at 532.
Hunk #4 failed at 558.
3 out of 4 hunks failed--saving rejects to plugins/crypto/openssl/ua_pki_openssl.c.rej
````

ccvca, 2021-07-16:
This should be resolved in the current master ( #4518 )

closed 2021-09-23 by jpfr

kjeldflarup, 2021-11-15:
I see this error on all 1.2x tags, the fix has not been released yet.  
Is there a way to compile using an official tag?

handsomeZWF, 2021-11-21:
> I see this error on all 1.2x tags, the fix has not been released yet. Is there a way to compile using an official tag?

How to resolve the problem?