open62541/open62541 #4215: Pub Sub fails after 1st successful publish to mosquitto mqtt

jackybek, 2021-03-01:
## Description

## Background Information / Reproduction Steps
OPC UA Pub Sub only works 1 time, after that, it returns the following error:

**PubSub MQTT: publish: Send buffer is full. Possible reasons: send buffer is to small, sending to fast, broker not responding.**

I have a OPC-UA server with PubSub capability.  I also run a separate agent program to read from a stream of input (XML-like) and populate the OPC-UA address space.
So long I run my OPC-UA server without running the agent program, the publish routine works.  However, when i start to populate the OPCServer using my agent program, the publish routine fails.

when i trace the source code, i saw the following:

1. Initial MQTT message content before it is populated with data extract from OPCUA node:
==============================================
open62541.c line 88730: bufsz = **1891**, fixed_header->remaining length = **1718**
application_message_size : 1704
Contents of the buffer is 
application_message : <<{"MessageId":null,"MessageType":"ua-data","Messages":[{"DataSetWriterId":2234,"SequenceNumber":0,"MetaDataVersion":{"MajorVersion":1043470538,"MinorVersion":1043457048},"Timestamp":"2021-03-01T21:39:54.256957Z","Payload":{"SoftwareVersion":{"Type":12,"Body":null},"DataBlockVersion":{"Type":12,"Body":null},"InstrumentTime":{"Type":12,"Body":null},"MeasurementTime":{"Type":12,"Body":null},"Sensor":{"Type":12,"Body":null},"OperatingTime":{"Type":4,"Body":0},"WarningMessage":{"Type":12,"Body":null},"BootStatus":{"Type":12,"Body":null},"SnapshotStatus":{"Type":12,"Body":null},"SCPStatus":{"Type":12,"Body":null},"SFTPStatus":{"Type":12,"Body":null},"RunscriptStatus":{"Type":12,"Body":null},"ArchiveStatus":{"Type":12,"Body":null},"AncillarySensorStatus":{"Type":12,"Body":null},"IgramPP":{"Type":10,"Body":0},"IgramDC":{"Type":10,"Body":0},"LaserPP":{"Type":10,"Body":0},"LaserDC":{"Type":10,"Body":0},"SingleBeamAt900":{"Type":10,"Body":0},"SingleBeamAt2500":{"Type":10,"Body":0},"SignalToNoiseAt2500":{"Type":4,"Body":0},"CenterBurstLocation":{"Type":10,"Body":0},"DetectorTemperature":{"Type":10,"Body":0},"LaserFrequency":{"Type":10,"Body":0},"HardDriveSpace":{"Type":4,"Body":0},"Flow":{"Type":10,"Body":0},"Temperature":{"Type":10,"Body":0},"Pressure":{"Type":10,"Body":0},"TempOptics":{"Type":10,"Body":0},"BadScanCounter":{"Type":4,"Body":0},"FreeMemorySpace":{"Type":4,"Body":0},"LABFilename":{"Type":12,"Body":null},"LOGFilename":{"Type":12,"Body":null},"LgFilename":{"Type":12,"Body":null},"SecondLgFilename":{"Type":12,"Body":null},"DetectorCounter":{"Type":10,"Body":0},"LaserCounter":{"Type":10,"Body":0},"FlowPumpCounter":{"Type":10,"Body":0},"DesiccantCounter":{"Type":10,"Body":0}}}]}>>
=====================================

2. When the message is populated with OPCUA Node:
=====================================
[2021-03-02 05:40:06.258 (UTC+0800)] info/server        publishMQTT(): before calling mqtt_publish() at 90118 : open62541.c :2185
open62541.c : mqtt_pac_publish_request() line 88927
start pos of buf: 2128088628
bufsz           : **149**
topic_name      : AirgardTopic
packet_id       : 60968
application_message_size : **2185**
application_message : <<{"MessageId":null,"MessageType":"ua-data","Messages":[{"DataSetWriterId":2234,"SequenceNumber":1,"MetaDataVersion":{"MajorVersion":1043470538,"MinorVersion":1043457048},"Timestamp":"2021-03-01T21:40:06.257227Z","Payload":{"SoftwareVersion":{"Type":12,"Body":"2.5.003000"},"DataBlockVersion":{"Type":12,"Body":"2.0.0"},"InstrumentTime":{"Type":12,"Body":"20210302053028"},"MeasurementTime":{"Type":12,"Body":"20210302053028"},"Sensor":{"Type":12,"Body":"Run"},"OperatingTime":{"Type":4,"Body":1},"WarningMessage":{"Type":12,"Body":""},"BootStatus":{"Type":12,"Body":"Complete"},"SnapshotStatus":{"Type":12,"Body":"Idle"},"SCPStatus":{"Type":12,"Body":"Idle"},"SFTPStatus":{"Type":12,"Body":"Idle"},"RunscriptStatus":{"Type":12,"Body":"Idle"},"ArchiveStatus":{"Type":12,"Body":"Disabled"},"AncillarySensorStatus":{"Type":12,"Body":""},"IgramPP":{"Type":10,"Body":11.0445995330810546875},"IgramDC":{"Type":10,"Body":5.785999774932861328125},"LaserPP":{"Type":10,"Body":25.0109996795654296875},"LaserDC":{"Type":10,"Body":10.8299999237060546875},"SingleBeamAt900":{"Type":10,"Body":13.621700286865234375},"SingleBeamAt2500":{"Type":10,"Body":15.97700023651123046875},"SignalToNoiseAt2500":{"Type":4,"Body":606},"CenterBurstLocation":{"Type":10,"Body":8189.84130859375},"DetectorTemperature":{"Type":10,"Body":116.84940338134765625},"LaserFrequency":{"Type":10,"Body":11740.5},"HardDriveSpace":{"Type":4,"Body":23658},"Flow":{"Type":10,"Body":2357.5},"Temperature":{"Type":10,"Body":50},"Pressure":{"Type":10,"Body":19.8840007781982421875},"TempOptics":{"Type":10,"Body":35},"BadScanCounter":{"Type":4,"Body":6},"FreeMemorySpace":{"Type":4,"Body":299},"LABFilename":{"Type":12,"Body":"[LabFiles]\\20210302\\03\\AG-012345678 FTIR DATA 20210302 033642.LAB"},"LOGFilename":{"Type":12,"Body":"[LogFiles]\\AG-012345678 AirGard Analysis results 2021-03-02.log"},"LgFilename":{"Type":12,"Body":"[LgFiles]\\2021\\03\\02\\Lg033642.txt"},"SecondLgFilename":{"Type":12,"Body":"[SecondLgFiles]\\2021\\03\\02\\secondLg033642.txt"},"DetectorCounter":{"Type":10,"Body":2358.5},"LaserCounter":{"Type":10,"Body":2362.5},"FlowPumpCounter":{"Type":10,"Body":0},"DesiccantCounter":{"Type":10,"Body":2354.5}}}]}Mpe▒
>>
........................................
1. remaining_length     : 14
2. remaining_length     : 2199
caller of mqtt_pack_fized_header is mqtt_pack_publish_request
In mqtt_pack_fixed_header function: 88685
mqtt_pack_fized_header section: bufsz = **149**
**88730: after do-while loop, bufsz = 146, fixed_header->remaining_length = 2199**
88731: checking if bufsz < fixed_header->remaining_length
**88977: something wenT wrong rv :  0**
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
=====================================
open62541.c : mqtt_pac_publish_request() line 88927
start pos of buf: 2128088628
bufsz           : 1976
topic_name      : AirgardTopic
packet_id       : 60968
application_message_size : 2185
application_message : <<{"MessageId":null,"MessageType":"ua-data","Messages":[{"DataSetWriterId":2234,"SequenceNumber":1,"MetaDataVersion":{"MajorVersion":1043470538,"MinorVersion":1043457048},"Timestamp":"2021-03-01T21:40:06.257227Z","Payload":{"SoftwareVersion":{"Type":12,"Body":"2.5.003000"},"DataBlockVersion":{"Type":12,"Body":"2.0.0"},"InstrumentTime":{"Type":12,"Body":"20210302053028"},"MeasurementTime":{"Type":12,"Body":"20210302053028"},"Sensor":{"Type":12,"Body":"Run"},"OperatingTime":{"Type":4,"Body":1},"WarningMessage":{"Type":12,"Body":""},"BootStatus":{"Type":12,"Body":"Complete"},"SnapshotStatus":{"Type":12,"Body":"Idle"},"SCPStatus":{"Type":12,"Body":"Idle"},"SFTPStatus":{"Type":12,"Body":"Idle"},"RunscriptStatus":{"Type":12,"Body":"Idle"},"ArchiveStatus":{"Type":12,"Body":"Disabled"},"AncillarySensorStatus":{"Type":12,"Body":""},"IgramPP":{"Type":10,"Body":11.0445995330810546875},"IgramDC":{"Type":10,"Body":5.785999774932861328125},"LaserPP":{"Type":10,"Body":25.0109996795654296875},"LaserDC":{"Type":10,"Body":10.8299999237060546875},"SingleBeamAt900":{"Type":10,"Body":13.621700286865234375},"SingleBeamAt2500":{"Type":10,"Body":15.97700023651123046875},"SignalToNoiseAt2500":{"Type":4,"Body":606},"CenterBurstLocation":{"Type":10,"Body":8189.84130859375},"DetectorTemperature":{"Type":10,"Body":116.84940338134765625},"LaserFrequency":{"Type":10,"Body":11740.5},"HardDriveSpace":{"Type":4,"Body":23658},"Flow":{"Type":10,"Body":2357.5},"Temperature":{"Type":10,"Body":50},"Pressure":{"Type":10,"Body":19.8840007781982421875},"TempOptics":{"Type":10,"Body":35},"BadScanCounter":{"Type":4,"Body":6},"FreeMemorySpace":{"Type":4,"Body":299},"LABFilename":{"Type":12,"Body":"[LabFiles]\\20210302\\03\\AG-012345678 FTIR DATA 20210302 033642.LAB"},"LOGFilename":{"Type":12,"Body":"[LogFiles]\\AG-012345678 AirGard Analysis results 2021-03-02.log"},"LgFilename":{"Type":12,"Body":"[LgFiles]\\2021\\03\\02\\Lg033642.txt"},"SecondLgFilename":{"Type":12,"Body":"[SecondLgFiles]\\2021\\03\\02\\secondLg033642.txt"},"DetectorCounter":{"Type":10,"Body":2358.5},"LaserCounter":{"Type":10,"Body":2362.5},"FlowPumpCounter":{"Type":10,"Body":0},"DesiccantCounter":{"Type":10,"Body":2354.5}}}]}Mpe▒
>>
........................................
1. remaining_length     : 14
2. remaining_length     : 2199
caller of mqtt_pack_fized_header is mqtt_pack_publish_request
In mqtt_pack_fixed_header function: 88685
[... cut ...]
Can you advise how to resolve this issue?

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.2-rc2-61-g0af95df8
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: raspbian linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

jackybek, 2021-03-10:
To put things in perspective, here is the mosquitto.conf :
per_listener_settings false
allow_zero_length_clientid true
#max_keepalive 65535
port 1883
max_connections 1024
protocol mqtt
allow_anonymous false
password_file /etc/mosquitto/conf.d/userlist.txt
**message_size_limit 1024000**

jackybek, 2021-03-15:
I did a trace of the code open62541.c and discovered why mqtt_pac_published_request() at line 88927 was called 2 times for every new payload. It is because of the 'MQTT_TRY_PACK() function : line 87846.  This function will try to pack the message .. if it fails the first time, it will try the second time.
So this explains observation number 1.

jackybek, 2021-04-04:
Issued is solved by initialising the following parameters of the structure:

    UA_KeyValuePair connectionOptions[3];
    connectionOptions[0].key = UA_QUALIFIEDNAME(0, "sendBufferSize");
    UA_UInt32 sBs = 1000;
    UA_Variant_setScalar(&connectionOptions[0].value, &sBs, &UA_TYPES[UA_TYPES_UINT32]);
    connectionOptions[1].key = UA_QUALIFIEDNAME(0, "recvBufferSize");
    UA_UInt32 rBs = 1000;
    UA_Variant_setScalar(&connectionOptions[1].value, &rBs, &UA_TYPES[UA_TYPES_UINT32]);

closed 2021-04-04 by jackybek

Deacychen257, 2022-08-11:
Thanks, the problem was solved!

NickBark, 2024-03-25:
Hi, faced the same issue v.1.3.9