open62541/open62541 #4216: Methods that are added under variable nodes cannot be called

smitha-cgi, 2021-03-01:
## Description
I have set up an OPC server which is modelling some SCADA data. Using the example of the pump in the [tutorial](https://open62541.org/doc/current/tutorial_server_object.html), I have set up a Status data variable node, and have added Start and Stop methods underneath it:

Device
  --> Status
       -----> Start()
       -----> Stop()

When I call the Start or Stop method I get a BadNodeClassInvalid status returned. This is because of a check performed at line 119 in [ua_services_method.c](https://github.com/open62541/open62541/blob/master/src/server/ua_services_method.c) where it checks the object head node class:

```
/* Verify the object's NodeClass */
    if(object->head.nodeClass != UA_NODECLASS_OBJECT &&
       object->head.nodeClass != UA_NODECLASS_OBJECTTYPE) {
        result->statusCode = UA_STATUSCODE_BADNODECLASSINVALID;
        return;
    }
```

I have commented out this check and the method can then be executed successfully. The question is: why does the head node class matter when calling a method? If I am allowed to add a method under a node with a class of UA_NODECLASS_VARIABLE without any errors shouldn't I be able to use it?
 
## Background Information / Reproduction Steps

Add a method under a data variable node and then try to call the method from a client. It will fail.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2021-03-02:
IMHO the OPC UA Spec Part 4, chapter 5.11.2 specifies that Objects shall be the parent node from a UA Method.
5.11.2.1
"Methods represent the function calls of Objects."

5.11.2.2 - Parameter objectId
"The NodeId shall be that of the Object or ObjectType on which the Method is invoked."

ccvca, 2021-03-02:
The object is also required for the OPC UA Client to call the Method:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.11.2/

smitha-cgi, 2021-03-18:
Having moved all of my methods to reside under an object (since that's what the spec seems to require) I'm happy enough to close this.

closed 2021-03-18 by smitha-cgi