open62541/open62541 #4222: Problem with a client reading custom datatypes

clemensg, 2021-03-03:
Hi,

I am evaluating open62541 for use in a client application that should read values from a Simatic S7 via OPC UA.

I read through the examples and tried to model the custom datatypes accordingly (similar to the custom_datatype example) but the data is not decoded as expected and I am not sure what I am doing wrong.

The custom datatype `DTL` is a structure with mostly `Byte` values (HOUR, MINUTE, DAY, etc.) but also a `UInt32` (NANOSECOND) and a `UInt16` (YEAR).
This is my datatype definition: https://gist.github.com/clemensg/b1857b19e726938170f8bc082d4cad27
And this is how I tried to read the value: https://gist.github.com/clemensg/d2c64619ff7af7c35bb33669e83c8dc5

However, something is going wrong, because the data in `dtl` which I am logging with `UA_LOG_INFO` is not correct.

And this is how it looks from a OPC UA browser (Prosys OPC UA Client):
DTL value: https://i.imgur.com/0sVeseo.png
NANOSECOND element of DTL: https://i.imgur.com/XRbDUkV.png

As I did not write the server code, how can I know the `.binaryEncodingId` of a custom datatype? I can see the nodeId of the type in a OPC UA browser under Types -> Data Types -> OPC Binary -> SimaticStructures -> DTL, there I see nodeId: `ns=3;s=TD_DTL`: https://i.imgur.com/TNiqGb9.png
So I used `UA_NODEID_STRING(3, "TD_DTL")` as `.typeId`, but what should the `.binaryEncodingId` in my `struct UA_DataType DTL` be?

Any help is greatly appreciated.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [x] Operating system: macOS & Windows

jpfr, 2021-03-07:
In OPC UA, every encoding (binary, JSON, XML) has a different NodeId.

A non-builtin datatype is wrapped in an ExtensionObject that also contains the NodeId of type+encoding.

Since you set the binaryId to ns=3;i=0 probably the decoding in the client could not find the type.

To get the correct binaryId you can either
- try if the Siemens PLC uses the "normal" type NodeId als for the encodings (which it shouldn't but could still do)
- look for the binary encoding nodeid in the server information model
- use a debugger with a break in the decoding code
- use Wireshark (which disects OPC UA perfectly) to see the encoding NodeId "on the wire"

jpfr, 2021-03-07:
And by the way, the DataType NodeId being different for different encodings, that is one of the biggest warts that OPC UA has.
But, in order to maintain backwards compatibility, it will probably not go away.

clemensg, 2021-03-07:
Hi, thank you for your help.

I used Wireshark to capture the ReadResponse message. This is the result: https://i.imgur.com/lJiyH6z.png

I noticed a different ExpandedNodeId Identifier string (TE_DTL instead of TD_DTL which is used for the .nodeId), so I tried `.binaryId = UA_NODEID_STRING(3, "TE_DTL")` and for the first time got back different values in the DTL struct but they are still not valid day, hour, year values.

Would this `UA_NODEID_STRING(3, "TE_DTL")` be the correct binaryId according to the capture screenshot?

ogsadmin, 2023-11-19:
Hi,

I had exactly the same issue and after searching through the issues here, the goggle group and the custom_datatype sample, I came up with (might not be the best way, but its working reliably) the following (see some sample code at the end of this post):
1. Read the nodeID of the `TE_<your type name>` type (once during startup)
2. For each write: create a new UA_ExtensionObject and initialize its encoding with your `TE_<your type name>` typeId
3. Copy it into a UA_Variant
4. Write it using UA_Client_writeValueAttribute()

Of course, you need to take care of the actual encoding yourself.

Please note, that the `TE_` prefix is something specific to Siemens PLCs - other PLCs don't require this or use other type names...

Unfortunately, the way how the UA_Client_Service_read() function parses the actual network response for an ExtensionObject (in Variant_decodeBinaryUnwrapExtensionObject()), drops the ExtensionObject typeId information (with each read response, the raw network data includes the `TE_<your type name>` typeID of the ExtensionObject. So far I did not find any way to read the `TE_<your type name>` typeId in any other way (mybe someone can help). Modifying the source to add a typeId node attribute to the UA_Variant class or (even better) changing the UA_ExtensionObject value accordingly would obviously the best solution (maybe somebody can file a issue and do a pull request...?)...

Here is what a read value from a Siemens structured data type returns (including the `TE_` prefix typeID for the ExtensionObject):
[image]

And here is the code from open62541 where it reads the typeId, but does not report it back to the caller (neither through the UA_ExtensionObject nor through the UA_Variant):
[image]

Anyway, here is some working sample code for writing to an ExtensionObject:
``` c++
UA_StatusCode TOpcUA_IOThread::writeExtensionObjectValue(const UA_NodeId nodeId, const UA_NodeId& dataTypeNodeId, const UA_ByteString *newValue)
{
	UA_Variant myVariant; // Variants can hold scalar values (ExtensionObject) and arrays of any type 
	UA_Variant_init(&myVariant);
	// build the extension object
	UA_ExtensionObject dataValue;
	dataValue.encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
	dataValue.content.encoded.typeId = dataTypeNodeId;
	dataValue.content.encoded.body.data = (UA_Byte*)newValue->data;
	dataValue.content.encoded.body.length = newValue->length;

	UA_Variant_setScalarCopy(&myVariant, &dataValue, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT]);
	if (myVariant.type == &UA_TYPES[UA_TYPES_STRING]) {
		// enforce bytestring (if not yet)
		myVariant.type = &UA_TYPES[UA_TYPES_BYTESTRING];
	}
	UA_StatusCode retval = UA_Client_writeValueAttribute(_client, nodeId, &myVariant);
	UA_Variant_clear(&myVariant);
	return retval;
}
```

closed 2026-10-20 by o6-nbeier