open62541/open62541 #4232: How can my OPC client connect to the server with X509 certificate authentication

zhiliangxiaoying, 2021-03-08:
such as uaexpert
I can see three ways to connect server
1，anonymous
2，username
3，certificate
but i read open62541 ，no certificate authentication examples were found;
So how can my OPC client connect to the server with X509 certificate authentication
thank u

zhiliangxiaoying, 2021-03-08:
[image]

[image]

juangburgos, 2021-03-09:
Take a look at [this PR](https://github.com/open62541/open62541/pull/3969)

MartinMajewski, 2021-04-01:
> such as uaexpert
> I can see three ways to connect server(prosys)
> 1，anonymous
> 2，username
> 3，certificate
> but i read open62541 ，no certificate authentication examples were found;
> So how can my OPC client connect to the server with X509 certificate authentication
> thank u

Would you mind pointing me to the section in the docs where those connection options are described? I'm missing something for sure as I am new to it, but I think the "clients" section should have a paragraph mentioning 3rd party clients.

NoelGraf, 2021-04-23:
You can find basic information in the documentation under Protocol/Establishing a Connection. More detailed information can also be found in the OPC UA documentation of the OPC foundation.

In the examples folder you can find some examples on this topic. For encryption (examples/encryption) but also for access control (examples/access_control) (user login).

NoelGraf, 2021-04-29:
I would close the issue for now. Please reopen if required!

closed 2021-04-29 by NoelGraf

zhiliangxiaoying, 2021-06-08:
> You can find basic information in the documentation under Protocol/Establishing a Connection. More detailed information can also be found in the OPC UA documentation of the OPC foundation.
>
> In the [...]

I've already read it；In the examples folder，just have username authentication and Certificate of encryption；
It's not about Certificate authentication；uaexpert have this function;
I dont know i should how to deal with it;

zhiliangxiaoying, 2021-06-08:
> I've already read it；In the examples folder，just have username authentication and Certificate of encryption；
> It's not about Certificate authentication；uaexpert have this function;
> I dont know i [...]

I can set up an OPCUA Server with certificate authentication function through OPC UA Simulation Server