open62541/open62541 #4234: How to connect a callback function to a method created with the Model Compiler?

klementuel, 2021-03-10:
Hello,
I would like to link my methods with a function, but despite a long search I can't find the solution.

In my Model C file, the Model Compiler has created methods, but they are not linked to the input and output arguments.

```
static UA_StatusCode function_MotorSteuerung_5_finish(UA_Server *server, UA_UInt16* ns) {
#ifdef UA_ENABLE_METHODCALLS
return UA_Server_addMethodNode_finish(server, 
UA_NODEID_NUMERIC(ns[1], 2013LU)
, NULL, 0, NULL, 0, NULL);
#else
return UA_STATUSCODE_GOOD;
#endif /* UA_ENABLE_METHODCALLS */
}
```
Now I thought I had to call UA_Server_addMethodNode_finish again in the main.c to set the corresponding arguments and functions. Unfortunately, I am already failing with the correct retrieval of the arguments from my model. However, here is the attempt to do this, maybe someone can tell me how to do this correctly?

```
UA_Variant *varInput = UA_Variant_new();
UA_Server_readValue(server, UA_NODEID_NUMERIC(nsIdx, 2014), varInput);
UA_Variant *varOutput = UA_Variant_new();
UA_Server_readValue(server, UA_NODEID_NUMERIC(nsIdx, 2015), varOutput);
UA_Argument *input = (UA_Argument*)varInput->data;
UA_Argument *output = (UA_Argument*)varOutput->data;
#ifdef UA_ENABLE_METHODCALLS
UA_Server_addMethodNode_finish(server, UA_NODEID_NUMERIC(nsIdx, 2012),
	                        &ControlMethodCallback,
	                        3, input,
	                        1, output);

#endif
```
I'm pretty sure that I'm reading the arguments incorrectly, as I only find part of the information in the input when debugging.

At the end I want to connect the whole thing with the following function, which should work, but I'm not quite sure here either...

```
static UA_StatusCode
ControlMethodCallback(UA_Server *server,
                         const UA_NodeId *sessionId, void *sessionHandle,
                         const UA_NodeId *methodId, void *methodContext,
                         const UA_NodeId *objectId, void *objectContext,
                         size_t inputSize, const UA_Variant *input,
                         size_t outputSize, UA_Variant *output) {
	//Handle Input 
	UA_Int16 speed_in = *(UA_Int16*)input[0].data;
	UA_Int16 distance_in = *(UA_Int16*)input[1].data;
	UA_Boolean direction_in = *(UA_Boolean*)input[2].data;
	
	//Handle Direction and set for test the return Value to direction value
	UA_Boolean ret_val = direction_in;
	UA_Variant value;
	UA_Variant_setScalar(&value, &ret_val, &UA_TYPES[UA_TYPES_BOOLEAN]);
	UA_Server_writeValue(server, UA_NODEID_NUMERIC(nsIdx, 2012), value);
	
	//Return the Value to Methode Output
    UA_Variant_setScalarCopy(output, &ret_val, &UA_TYPES[UA_TYPES_BOOLEAN]);
	
	//End Methode
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Control was called");
    return UA_STATUSCODE_GOOD;
}
```
If my example works, after your help probably, you could possibly include this in the examples, here I found no examples where it is described how to modify the code if you used the model compiler before.

Many thanks in advance and best regards
Klemens

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [x] Other OPC UA SDKs used (client or server): UA_Expert
 - [x] Operating system: Linux Debian as VM Finale Version for Raspberry
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

klementuel, 2021-03-12:
sorry it was my fault
I don´t know why, but I used the wrong Node ID. As you can see in the code above, I used 2012 instead of 2013.

now I can set the callback with this simple function
```
//Set Callback for M1 Control https://open62541.org/doc/1.2/server.html#method-callbacks
UA_Server_setMethodNode_callback(server,
                            	UA_NODEID_NUMERIC(nsIdx, 2013),
                            	   &ControlMethodCallback);
```

closed 2021-03-12 by klementuel

shaohuazhang1, 2022-08-14:
where and how you declare variable nsIdx?