open62541/open62541 #4236: Doesn't build with MQTT option enabled

fatadel, 2021-03-10:
## Description

I have installed the library according to this [tutorial](https://open62541.org/doc/current/installing.html#manual-installation).
It builds just fine with the options given in the tutorial (`-DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL`). It also builds fine with `-DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON` options. However, whenever I add `-DUA_ENABLE_PUBSUB_MQTT=ON` the build process fails.

The output of cmake gives me many times the following error:

```
%SOURCE-LOCATION%/open62541/deps/mqtt-c/mqtt.c:38:5: error: implicit declaration of function 'pthread_mutex_lock' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
    MQTT_PAL_MUTEX_LOCK(&client->mutex);
    ^
%SOURCE-LOCATION%/open62541/deps/mqtt-c/mqtt_pal.h:76:42: note: expanded from macro 'MQTT_PAL_MUTEX_LOCK'
    #define MQTT_PAL_MUTEX_LOCK(mtx_ptr) pthread_mutex_lock(mtx_ptr)
```

And then eventually the build process fails with error:

```
fatal error: too many errors emitted, stopping now [-ferror-limit=]
```

## Background Information / Reproduction Steps

1. Install the library according to the [tutorial](https://open62541.org/doc/current/installing.html#manual-installation) but additionally add `-DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_MQTT=ON` to the cmake options. So that, you have:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_MQTT=ON ..
```

Used CMake options:

Described above.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2-68-gdbc4cc1a
 - [x] Other OPC UA SDKs used (client or server): no
 - [x] Operating system: macOS Big Sur. It's worth mentioning that I am building on an M1 (arm) Apple Silicon chip.
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-03-13:
Works without a hitch on Linux. Possibly Mac related.
From the error messages it looks as if you miss an include for `pthread_mutex_lock`.

NoelGraf, 2021-04-08:
What is the status? Does the problem still exist?

closed 2021-04-16 by NoelGraf

fatadel, 2021-04-16:
> What is the status? Does the problem still exist?

@NoelGraf 
Sorry that I haven't responded.

Yeah, the issue still exists. We have also checked on another mac (Intel-based).