open62541/open62541 #4243: Binaries for Release 1.2 are missing

JensMertelmeyer, 2021-03-15:
The [open62541/releases/tag/v1.2](https://github.com/open62541/open62541/releases/tag/v1.2) only lists a file for Debian. The file itself is just a text file containing "Test file1".

Links on https://open62541.org/ are broken as well.

JensMertelmeyer, 2021-04-21:
Still the same with 1.2.1, binaries are all missing.

Maybe a short animation makes it more clear? 

https://user-images.githubusercontent.com/1827784/115514743-b7ebd180-a284-11eb-94fd-4bae6c88aea5.mp4

vranki, 2023-02-08:
Still no binaries to be found for latest release, v1.3.4

closed 2025-02-04 by andreasebner

JensMertelmeyer, 2025-02-05:
Some kind of comment would have been nice.

The homepage still states "You can always find prebuild binaries for every release on our [Github Release Page.](https://github.com/open62541/open62541/releases)" which, in fact, does not offer any kind of binaries anymore 🤷‍♂️