open62541/open62541 #4245: How to get element index when use DataSource reading array?

jixian79, 2021-03-16:
If I create a variable node with bool array datatype, and i use datasrouce to read the data from a modbus device to the array. When i change one of the elements of the array ,  how can i get the element index from the datasource callback function?   is there any way to only write the changed value to the physical device?

in the datasource function signature, the parameter related with the variable is  `UA_DataValue* data `， but it seems have no information about the index.

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2
 - [x] Other OPC UA SDKs used (client or server): UaExpert
 - [x] Operating system:Windows

jpfr, 2021-03-16:
See here: https://open62541.org/doc/current/nodestore.html#variablenode

The DataSource callback contains the NumericRange attribute.

jixian79, 2021-03-25:
> See here: https://open62541.org/doc/current/nodestore.html#variablenode
> 
> The DataSource callback contains the NumericRange attribute.

[image]

But actually, the NumericRange is empty while running the write callback。

[image]

I Use the UaExpert change the 9th bool。In debug mode, the NumericRange is a null pointer.

jixian79, 2021-03-25:
> See here: https://open62541.org/doc/current/nodestore.html#variablenode
> 
> The DataSource callback contains the NumericRange attribute.

[image]

But actually, the NumericRange is empty while running the write callback。

[image]

I Use the UaExpert change the 9th bool。In debug mode, the NumericRange is a null pointer.

So how can i get the index information from the write callback ?

mlang-de, 2021-03-25:
The client has to send the NumericRange. I guess the UaExpert does not support writing with ranges.
May check it with Wireshark.

jixian79, 2021-03-26:
thank you very much

NoelGraf, 2021-04-08:
Has your question been clarified? Can the issue be closed?

closed 2021-04-15 by NoelGraf