open62541/open62541 #4248: using open62541 with remote mqtt broker

Sarutobie, 2021-03-16:
## Description

As far as i understand the mqtt publisher example only works with a local broker. Is there a possiblity to use a remote broker or is it not possible atm?

Used CMake options:

```
UA_ENABLE_PUBSUB_MQTT, UA_ENABLE_PUBSUB_ETH_UADP, UA_ENABLE_PUBSUB
```

Another question is if i can publish on mqtt and uadp simultaneously?
In my test i have added transportlayer for uadp and mqtt. but it is only being published on the last added layer.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-03-18:
>  In my test i have added transportlayer for uadp and mqtt. but it is only being published on the last added layer.

That sounds like an issue.

jackybek, 2021-03-27:
I can publish to a remote mqtt server using a simple network switch :

opcua server -> mosquitto mqtt server

jackybek, 2021-03-30:
i can connect my OPCUA Server running on 1 node and publish to a MQTT server running on another node remotely on the networt.  At the same time, i can also using wireshark to observe the traffic using UADP, simultaneously.  But the MQTT Publish routine currently has a bug (#4215), such that it only allows for 1 cycle of publishing, the next cycle will end up having an error.

Rafuxy, 2021-07-06:
@jackybek  Please can you tell me if your Brocker (MQTT Server) run with key (TLS)? If yes how do to programe do make sure that the brocker your OPCUA enable the connection with the given key?

jackybek, 2021-07-06:
Not yet .. I just upgraded my opcua binary to v1.2.1 (which has the ability for opcua server to send data over encrypted mqtt transport).  I will try out the new code base soon.

But theoretically it should work this way, treating the mqtt broker as a server and opcua server as a client, and communicate via OpenSSL cert and public key.   It should be similar to opcua client communicating with opcua server over secure channel via OpenSSL cert and public key, which I have already completed.

closed 2026-10-20 by o6-nbeier