open62541/open62541 #4258: Unable to create monitored item when value status code is 'bad'

GrantEdwards, 2021-03-18:
## Description

The open62541 server will not create a monitored item if the item's value's status code is bad at the time of the request.

## Background Information / Reproduction Steps

A number of my process-data values have status codes that can be temporarily 'bad' for various reasons (sensor disconnected, value out of range, value marked invalid by user, etc). When values are read in these states, my application code sets the value's status code to various BAD values depending on the situation.

Once a monitored item is created, everything works fine: the client is notified when the value or status changes.

But if the client tries to create a monitored item at a point in time when the value has a BAD status code, the open62541 will refuse to create the monitored item and returns the value's BAD status code.

I don't understand this. Why does the value's status code have any bearing on whether you can monitor it or not?

Are BAD status codes assumed to be permanent and never change to GOOD?

Even if the status remains BAD, the value can still change and can still be useful information.

Used CMake options:

~~~
-- The C compiler identification is GNU 4.6.3
-- The CXX compiler identification is GNU 4.6.3
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/nextgen/toolchain/bin/arm-linux-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/nextgen/toolchain/bin/arm-linux-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python (found version "3.8.8") 
-- Found Git: /usr/bin/git (found version "2.26.2") 
-- open62541 Version: 0.0.01.0.1
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- The selected architecture is: posix
-- Found MBEDTLS: /home/nextgen/opcua/mbedtls-2.16.2/include  
-- Performing Test CC_HAS_-fstack-protector-strong
-- Performing Test CC_HAS_-fstack-protector-strong - Failed
-- Performing Test CC_HAS_-fstack-clash-protection
-- Performing Test CC_HAS_-fstack-clash-protection - Failed
-- Performing Test CC_HAS_-mcet -fcf-protection
-- Performing Test CC_HAS_-mcet -fcf-protection - Failed
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nextgen/opcua/open62541-build
-- Cache values
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_TOOLCHAIN_FILE:FILEPATH=/home/nextgen/opcua/Toolchain-arm-linux.cmake
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=REDUCED

~~~

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0.1
 - [x] Other OPC UA SDKs used (client or server): none
 - [x] Operating system: Linux on ARM9
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

GrantEdwards, 2021-03-18:
I've rebuilt with UA_LOGLEVEL==0, and the only relevant log messages I can find are

~~~
[1194.507]   debug/session  Connection 25 | SecureChannel 1 | Session ns=1;g=87d26c6d-1959-f10d-5335-7387c73c23c2 | Processing CreateMonitoredItemsRequest
[1194.509]   debug/session  Connection 25 | SecureChannel 1 | Session ns=1;g=87d26c6d-1959-f10d-5335-7387c73c23c2 | Read the attribute 13
~~~

GrantEdwards, 2021-03-18:
Somebody on the mailing list has already suggested using 'uncertain' instead of 'bad' for the value's statuscode when the value is invalid. Doing that would violate §7.1.3 of the IO-Link Foundation's specification "OPC Unified Architecture for IO-Link", which explicitly requires that the statuscode for invalid process data be a 'bad' statuscode.

jpfr, 2021-03-18:
Make sense. The fix is in #4259.
Please comment if you are not okay with the remaining StatusCodes that lead to an abort of AddMonitoredItems.

GrantEdwards, 2021-03-18:
Thanks!

That was fast. I was about to start working on a patch, but it
probably would have taken me half a day to figure out where/what to
change.

I haven't tested the change, but it certainly looks like it will work
for me.

However, it looks like Bad_UserAccessDenied will still cause an abort,
and that's one of the two status codes that the spec specifically says
should not cause a failure:

From part 04 of the spec

> 5.12.2 CreateMonitoredItems
>
> 5.12.2.1 Description
>
>[...]
>
> When a user adds a monitored item that the user is denied read
> access to, the add operation for the item shall succeed and the bad
> status Bad_NotReadable or Bad_UserAccessDenied shall be returned in
> the Publish response.

Do you need to swap UA_STATUSCODE_BADINDEXRANGENODATA and UA_STATUSCODE_BADUSERACCESSDENIED in your change?

GrantEdwards, 2021-03-19:
I back-ported the changes below to my v1.0.1 sources and that fixed my problem.

 * [pull request 4259](https://github.com/open62541/open62541/pull/4259/)
 * [pull request 4260](https://github.com/open62541/open62541/pull/4260/)

I'm not sure what your workflow is regarding issue closure, but I'm ok with closing this one.

closed 2021-03-19 by jpfr