open62541/open62541 #4262: BadEncodingLimitsExceeded

bavilla, 2021-03-20:
Hello,
Connecting to the server opc.tcp://195.182.154.240:7718 throws an error:

[2021-03-20 17:16:15.748 (UTC+0300)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-03-20 17:16:15.749 (UTC+0300)] trace/client	Client connect iterate
[2021-03-20 17:16:15.759 (UTC+0300)] trace/client	Client connect iterate
[2021-03-20 17:16:15.759 (UTC+0300)] debug/network	Sent HEL message
[2021-03-20 17:16:15.759 (UTC+0300)] debug/client	Client Status: ChannelState: HELSent, SessionState: Closed, ConnectStatus: Good
[2021-03-20 17:16:15.759 (UTC+0300)] trace/client	Client connect iterate
[2021-03-20 17:16:15.768 (UTC+0300)] error/channel	Connection 84 | SecureChannel 0 | Received an ERR response with StatusCode BadEncodingLimitsExceeded and the following reason: TL error 0x80080000
[2021-03-20 17:16:15.768 (UTC+0300)] warn/channel	Connection 84 | SecureChannel 0 | Receiving the response failed with StatusCode Good
[2021-03-20 17:16:15.768 (UTC+0300)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed

UAExpert connects and works with this server. What could be the reason?

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-03-21:
You have to give enough information to reproduce.

- Open62541 is used for the client?
- Encryption is used?
- What is the remote server?

bavilla, 2021-03-21:
> You have to give enough information to reproduce.
> 
> * Open62541 is used for the client? Yes
> * Encryption is used? No
> * What is the remote server? Eldis

I deliberately opened access to this server so that someone from the developers could try to connect

bavilla, 2021-03-22:
Julius, maybe you need to know the implementation details of the Eldis-OPC server? I can contact these developers.

bavilla, 2021-04-11:
The problem was on the side of the server developers. The issue is resolved.

closed 2021-04-11 by bavilla