open62541/open62541 #4279: Is there a difference between open62541 OPC UA and classic OPC UA

MichaelKlin, 2021-03-24:
## Description
Or has the open62541 the same functionality as classic OPC UA from OPC Foundation. I cannot really found the difference. Thanks in advance
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2021-03-24:
Imho there are OPC UA and OPC classic.
OPC classic is the "first version" of OPC and uses the Windows-based DCOM interface for communication.
OPC UA is the new one (since i guess 2009 released OPC UA Spec.).
OPC UA is platform indepentend, OPC Classic is only for Windows.
OPC UA is also available as norm IEC-62541.
The open62541 open source project realize/implemente the OPC UA Specification form the OPC Foundation (Consequently also the IEC62541).

MichaelKlin, 2021-03-24:
@mlang-de i understand difference OPC UA and OPC classic. I want only ask, if OPC UA Specification form the OPC Foundation are implemented. Or are missing some features/functions?

mlang-de, 2021-03-24:
`I want only ask, if all OPC UA Specifications form the OPC Foundation are implemented.`

Yes.

`For example mqtt is not from OPC UA Specification`

Why is it not from OPC UA Spec.? Part 14 for PubSub defines this.

MichaelKlin, 2021-03-24:
@mlang-de you are right. Part 14 is for mqtt. Thank you

mlang-de, 2021-03-24:
`if all OPC UA Specifications form the OPC Foundation are implemented`
To be honest, not ALL from OPC UA specifications are implemented.
But the parts which are implemented meets the needs for the OPC UA specification.

If this issue is solved for you, please close this issue. Tia

MichaelKlin, 2021-03-24:
@mlang-de i write bachelor thesis at the University and one part of my work is to compare OPC UA open source implementation with OPC UA specification. Thats why i want to ask

EvgKlin, 2021-03-29:
@mlang-de sorry for interrupting, i found out, that some sevices and security token is not supported. A small theoretical question, if it is ok. I  thinked many services like browsing of nodes is hapenning on client side, but in function call is using server pointer like this:
`void Service_Browse(UA_Server *server, UA_Session *session,const UA_BrowseRequest *request,UA_BrowseResponse *response); `

mlang-de, 2021-03-29:
Client-Server model uses request/response pattern. Therefore an OPC UA Server shall support the Browse Service too.
Other way around, how should the client browse, if the server does not support the service?

EvgKlin, 2021-03-29:
@mlang-de ok, until now i used only some functions like `UA_Client_addVariableNode` or something from your tutorials and there was not  request/response pattern. than you mean it only activate this service for the client (and it is void function)

mlang-de, 2021-03-29:
I'm no client expert furthermore these are the open62541 tutorials not mine.
I'm not sure what your goal is.
The named function UA_Client_addVariableNode is an indirect function call from UA_Client_Service_addNodes.
This means the Client uses AddNode-Service defined in Part 4.
Do you want to create a new Namespace/AddressSpace within the server?

Normally the server provide its address space an the client can "use" them.
Imho the intention is not to create the Server Address Space by the AddNode-Service since the is no information for the server where the real data shall be gathered from.

EvgKlin, 2021-03-29:
I use this function for adding Nodes to the Server (Node Managment). On my server i added AddressRaum 3, because it was necessary to add it (for my job). I have never used `UA_Client_Service_addNodes` and do not really know, how i would use  UA_RequestHeader. But I found some examples in ua_client_highlevel.c. My target/goal is to learn functions/passibility of OPC UA and to learn open62541.

mlang-de, 2021-03-29:
Again, the Node Management UA Service are imho not the recommended way to fill up the server address space!
The server shall be provide the namespaces with node (object, variables, methods...) and so its server address space by the Server application und NOT via the client. But this could be very use-case-dependend. Use the public Server API like UA_Server_addVariableNode and so on.
May check https://github.com/open62541/open62541/blob/master/examples/server_ctt.c

Don't get me wrong, but maybe https://www.buecher.de/shop/netzwerk-und-sicherheit/opc-unified-architecture/mahnke-wolfgangleitner-stefan-helmutdamm-matthias/products_products/detail/prod_id/23811103/ could help you to learn and understand what OPC UA is.

EvgKlin, 2021-03-30:
@mlang-de I used that 
https://www.amazon.de/OPC-englischsprachige-Ausgabe-Unified-Architecture/dp/3800732424
but sturcture of the book is terrible and sometimes i am confused. Your book looks nice

mlang-de, 2021-03-30:
I never read the book you post, but i guess this is more for OPC Classic. (Only at the end is some little OPC UA part)
The book I send you is good for server-client, I gues PubSub was not really introduced.
Furthermore there is the book https://www.amazon.de/Praxishandbuch-OPC-Implementierung-Nachr%C3%BCstung-Praxisbeispiele/dp/3834334545/ref=sr_1_5?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=opc+ua&qid=1617087238&sr=8-5 which could be a good starting point to see what could be archived with OPC UA.

EvgKlin, 2021-03-30:
@mlang-de =)) I order this book at university library, I get it only in April. And your first book is not at the library, i will search, where can i get it without big costs. If it is ok, i close then this issue

closed 2021-03-31 by EvgKlin