open62541/open62541 #4281: How to change and control AddressSpace in real time

EvgKlin, 2021-03-25:
## Description
I read all OPC UA Specification and i get some questions: How do i change and control AddressSpace in real time:
I know only this functions: UA_Server_addNamespace(server, "Number");
But i am not really sure, if it is right.
And second question is by using Types of AddressSpace-Concept can i export it to the client (in German: Das auf Types basierende AddressSpace-Konzept kann als Grundlage für die Veröffentlichung aller erförderlichen Informationen verwendet werden)  
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2021-03-26:
```
How do i change and control AddressSpace in real time:
I know only this functions: UA_Server_addNamespace(server, "Number");
```
Please check the provided examples to see how to handle an namespace and its content like Variables, Objects or Methods.

```
And second question is by using Types of AddressSpace-Concept can i export it to the client (in German: Das auf Types basierende AddressSpace-Konzept kann als Grundlage für die Veröffentlichung aller erförderlichen Informationen verwendet werden)
```

A client can browse a server address space and so the client can read the supported server DataTypes as well. Not sure what you want to know with your question.

EvgKlin, 2021-03-29:
@mlang-de Thank you, now i understanding

EvgKlin, 2021-03-29:
i close this issue

closed 2021-03-29 by EvgKlin