open62541/open62541 #4287: Events options : require FULL ua_namespace_zero : code breaks when adding new variables to OPCAddressSpace

jackybek, 2021-03-29:
## Description
[FULLNamesSpaceGDB with 200 log.txt](https://github.com/open62541/open62541/files/6219119/FULLNamesSpaceGDB.with.200.log.txt)

## Background Information / Reproduction Steps

Hi,
I am trying to implement Events and during the ccmake process, it says that I need to:
1. enable UA_NAMESPACE_ZERO
2. set to FULL

After I enabled this option, all the UA_Server_addVariableNode statements fail at runtime:
Error : segmentation fault.  See example below.

UA_UInt16 namespaceIndex=1;

UA_Server_addVariableNode(uaServer, UA_NODEID_NUMERIC(namespaceIndex, 10001), 
            r1_airgardroot_Id,                          
            UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY), 
            UA_QUALIFIEDNAME(namespaceIndex, "Software Version"),  
            UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
            vSoftwareVersionAttr, NULL, NULL);

What pre-requisites do I need to perform before enabling FULL ua_namespace_zero?
Is there any documentations to guide me?
I saw 2 related options in ccmake:
1. UA_FILE_NS0
2. UA_FILE_NS0_BLACKLIST

Do I need to set to 'ON'?
Thanks.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```
BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
BIBTEX_COMPILER:FILEPATH=BIBTEX_COMPILER-NOTFOUND
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUND
CLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUND
CMAKE_AR:FILEPATH=/usr/bin/ar
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_COLOR_MAKEFILE:BOOL=ON
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8
CMAKE_C_FLAGS:STRING=
CMAKE_C_FLAGS_DEBUG:STRING=-g
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_EXE_LINKER_FLAGS:STRING=
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFF
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
CMAKE_INSTALL_INCLUDEDIR:PATH=include
CMAKE_INSTALL_INFODIR:PATH=
CMAKE_INSTALL_LIBDIR:PATH=lib
CMAKE_INSTALL_LIBEXECDIR:PATH=libexec
CMAKE_INSTALL_LOCALEDIR:PATH=
CMAKE_INSTALL_LOCALSTATEDIR:PATH=var
CMAKE_INSTALL_MANDIR:PATH=
CMAKE_INSTALL_OLDINCLUDEDIR:PATH=/usr/include
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_INSTALL_RUNSTATEDIR:PATH=
CMAKE_INSTALL_SBINDIR:PATH=sbin
CMAKE_INSTALL_SHAREDSTATEDIR:PATH=com
CMAKE_INSTALL_SYSCONFDIR:PATH=etc
CMAKE_LINKER:FILEPATH=/usr/bin/ld
CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/make
CMAKE_MODULE_LINKER_FLAGS:STRING=
CMAKE_MODULE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_NM:FILEPATH=/usr/bin/nm
CMAKE_OBJCOPY:FILEPATH=/usr/bin/objcopy
CMAKE_OBJDUMP:FILEPATH=/usr/bin/objdump
CMAKE_RANLIB:FILEPATH=/usr/bin/ranlib
CMAKE_SHARED_LINKER_FLAGS:STRING=
CMAKE_SHARED_LINKER_FLAGS_DEBUG:STRING=
CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELEASE:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
CMAKE_SKIP_RPATH:BOOL=NO
CMAKE_STATIC_LINKER_FLAGS:STRING=
CMAKE_STATIC_LINKER_FLAGS_DEBUG:STRING=
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELEASE:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_STRIP:FILEPATH=/usr/bin/strip
CMAKE_VERBOSE_MAKEFILE:BOOL=FALSE
CPACK_SOURCE_RPM:BOOL=OFF
CPACK_SOURCE_TBZ2:BOOL=ON
CPACK_SOURCE_TGZ:BOOL=ON
CPACK_SOURCE_TXZ:BOOL=ON
CPACK_SOURCE_TZ:BOOL=ON
CPACK_SOURCE_ZIP:BOOL=OFF
DVIPDF_CONVERTER:FILEPATH=/usr/bin/dvipdf
DVIPS_CONVERTER:FILEPATH=DVIPS_CONVERTER-NOTFOUND
GIT_EXECUTABLE:FILEPATH=/usr/bin/git
HTLATEX_COMPILER:FILEPATH=HTLATEX_COMPILER-NOTFOUND
LATEX2HTML_CONVERTER:FILEPATH=LATEX2HTML_CONVERTER-NOTFOUND
LATEX_COMPILER:FILEPATH=LATEX_COMPILER-NOTFOUND
LUALATEX_COMPILER:FILEPATH=LUALATEX_COMPILER-NOTFOUND
MAKEINDEX_COMPILER:FILEPATH=MAKEINDEX_COMPILER-NOTFOUND
MDNSD_LOGLEVEL:STRING=300
OPENSSL_CRYPTO_LIBRARY:FILEPATH=/usr/local/ssl/lib/libcrypto.so
OPENSSL_INCLUDE_DIR:PATH=/usr/local/ssl/include
OPENSSL_SSL_LIBRARY:FILEPATH=/usr/local/ssl/lib/libssl.so
PDFLATEX_COMPILER:FILEPATH=PDFLATEX_COMPILER-NOTFOUND
PDFTOPS_CONVERTER:FILEPATH=/usr/bin/pdftops
PKG_CONFIG_EXECUTABLE:FILEPATH=/usr/bin/pkg-config
PS2PDF_CONVERTER:FILEPATH=/usr/bin/ps2pdf14
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python
SPHINX_EXECUTABLE:FILEPATH=/usr/bin/sphinx-build
UA_AMALGAMATION_ARCHITECTURES:STRING=
UA_ARCHITECTURE:STRING=posix
UA_ARCH_ADD_FLAGS:STRING=
UA_ARCH_EXTRA_INCLUDES:STRING=
UA_ARCH_LINKER_FLAGS:STRING=
UA_ARCH_REMOVE_FLAGS:STRING=
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_FUZZING:BOOL=OFF
UA_BUILD_FUZZING_CORPUS:BOOL=OFF
UA_BUILD_OSS_FUZZ:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_DEBUG:BOOL=OFF
UA_DEBUG_DUMP_PKGS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_CLANG_COV:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DETERMINISTIC_RNG:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=ON
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_IMMUTABLE_NODES:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_MALLOC_SINGLETON:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=ON
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=ON
UA_ENABLE_PUBSUB_ETH_UADP_XDP:BOOL=OFF
UA_ENABLE_PUBSUB_FILE_CONFIG:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=ON
UA_ENABLE_PUBSUB_MONITORING:BOOL=ON
UA_ENABLE_PUBSUB_MQTT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
**UA_ENABLE_SUBSCRIPTIONS:BOOL=ON**
**UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=ON**
**UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON**
UA_ENABLE_TYPEDESCRIPTION:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_FILE_NS0:BOOL=OFF
UA_FILE_NS0_BLACKLIST:BOOL=OFF
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_CPP:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=0
**UA_NAMESPACE_ZERO:STRING=FULL**
UA_PACK_DEBIAN:BOOL=OFF
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND

## Checklist

Please provide the following information:

 - [ X] open62541 Version (release number or git tag): v1.2-rc2-61-g0af95df8
 - [ X] Other OPC UA SDKs used (client or server): libxml2, mariadbclient, json-c, openssl
 - [ X] Operating system: raspbian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ X] Critical issue

EvgKlin, 2021-04-09:
Do you know, what UA_NAMESPACE_ZERO do exactly?

jackybek, 2021-04-13:
Not exactly..  but I was using minimal namespace zero all this while and it meets my needs so far, until I tried to implement alarms and conditions.  It requires full namespace zero.

jackybek, 2021-07-17:
Issues resolved due to invalid spaces in open62541.h and open62541.c.