open62541/open62541 #4296: (PubSub) Question regarding PubSubChannel->regist()

Fischer-Thomas, 2021-04-01:
## Description
As far as i can see from the different implementations of the PubSubChannel regist method (UDP, ETH, MQTT) i understood that the regist() implements functionality for the subscriber.
However when you add a publisher to the UAServer it performs the regist function. For a UDP publisher this doesn't cause any issues but for a MQTT publisher we will run into an error. 
I suggest to remove this lines of code https://github.com/open62541/open62541/blob/be65121bedd3a02b68e2124bd204f6d4759a0b09/src/pubsub/ua_pubsub_writer.c#L170-L174 unless there is something i have missed why there is a need to perform a PubSubChannel->regist() for a publisher.

What do you think? @jpfr @andreasebner 

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag):master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

MaoniinoaM, 2021-04-07:
Hi, @jpfr @andreasebner ,
I try to test PUBSUB MQTT. And I referred to the code: https://github.com/open62541/open62541/blob/master/examples/pubsub/tutorial_pubsub_mqtt_publish.c
It always return the error UA_STATUSCODE_BADARGUMENTSMISSING.
I review the code and the error path is as below:
UA_Server_addWriterGroup -> UA_PubSubConnection_regist -> connection->channel->regist(UA_PubSubChannelMQTT_regist) 
In UA_PubSubConnection_regist, it invoke UA_PubSubChannelMQTT_regist(connection->channel, NULL, NULL). But in UA_PubSubChannelMQTT_regist, it will check the transportSettings and should not be NULL.
So it seems a bug here obviously.

opcua-tsn-team-kalycito, 2021-04-07:
Hi @Fischer-Thomas,
We are working on this issue on this PR [#4235](https://github.com/open62541/open62541/pull/4235)

closed 2021-04-07 by Fischer-Thomas