open62541/open62541 #4303: Server not publishing dataChangeNotification if new value is a null variant

simonyg, 2021-04-06:
## Description

## Background Information / Reproduction Steps

DataNotification is not sent for data with error status change. Value is default empty UA_Variant.

Server code for publishing the change:

```
  UA_WriteValue wv;
  
  UA_WriteValue_init(&wv);
  wv.nodeId = nodeId;
  wv.attributeId = UA_ATTRIBUTEID_VALUE;
  wv.value.sourceTimestamp = util::toUADateTime(dateTime);
  wv.value.hasSourceTimestamp = true;
  wv.value.status = UA_STATUSCODE_BADNOTCONNECTED;
  wv.value.hasStatus = true;
  VerifyReturn(UA_Server_write(m_uaServer, &wv));
```

Client code for monitoring the item:

```
  UA_DataChangeFilter *dataChangeFilter = UA_DataChangeFilter_new();
  dataChangeFilter->trigger = UA_DATACHANGETRIGGER_STATUSVALUETIMESTAMP;
  dataChangeFilter->deadbandType = UA_DEADBANDTYPE_NONE
```

Work in v1.1.x but is broken in v.1.2.

Used CMake options:

```
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo  -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS=ON -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: MacOS, Ubuntu, Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

simonyg, 2021-04-06:
[image]

jpfr, 2021-04-07:
Thanks for figuring out the underyling issue. 👍 
We have reached to absolute lower bound for fix-sizes.
The one-character fix is in #4309 together with a new unit test to prevent regressions.

PR #4309 (2021-04-07): fix(server): Correct MonitoredItem filter for empty values
Fixes #4303.
commit 5a99a72e (2021-04-07): fix(server): Correct MonitoredItem filter for empty values

Fixes #4303.
closed 2021-04-07 by jpfr