open62541/open62541 #4310: size of VaribaleNode

embbo, 2021-04-07:
Hi ALL I am looking to find the size of varibaleNode ? how much heap take by each node.?

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-04-07:
(gdb) print sizeof(UA_VariableNode)
$1 = 440

(gdb) print sizeof(UA_ReferenceTarget)
$6 = 104

You need at least four references (to the parent and to the VariableType, and back from them).
With some additional smaller overhead I would say you need about 1kB RAM per VariableNode plus the value itself.

This can be significantly decreased by a specialized Nodestore that compresses and only keeps active nodes in RAM.
There is a first effort in that direction here: #3411

embbo, 2021-04-08:
@jpfr okay , So what i understand is 
1..UA_VariableNode take 1kb + some bytes
How much this function take  

 UA_ServerConfig_setDefaultWithSecurityPolicies(uaServerConfig,
			 portNumber,
			 &certificate,     &privateKey,
			 trustList,        trustListSize,
			 issuerList,       issuerListSize,
			 revocationList,   revocationListSize);
			 
			 because what I find this function eat most of the RAM ? and I can creat less Nodes (UA_VariableNode )

jpfr, 2021-04-08:
You are better off using a debugger to find out these numbers yourself.
That will help you go into arbitrary detail when optimizing for your usage scenarios.

closed 2021-04-08 by jpfr