open62541/open62541 #4313: AddNode (ns=1;i=10005): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.

Demolishun, 2021-04-07:
## Description
Getting this runtime message:
AddNode (ns=1;i=10005): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.

## Background Information / Reproduction Steps
`UA_VariableAttributes vAttr = UA_VariableAttributes_default;
    vAttr.description = UA_LOCALIZEDTEXT(localizedEng, desc.data());
    vAttr.displayName = UA_LOCALIZEDTEXT(localizedEng, qname.data());
    vAttr.dataType = UA_TYPES[UA_TYPES_FLOAT].typeId;
    vAttr.valueRank = UA_VALUERANK_SCALAR;
    vAttr.accessLevel = UA_ACCESSLEVELMASK_READ;
    UA_Float floatVar = point.read();
    UA_Variant_setScalar(&vAttr.value, &floatVar, &UA_TYPES[UA_TYPES_FLOAT]);
    UA_DataSource paramDataSource;
    paramDataSource.read = readItemValue;
    paramDataSource.write = writeItemValue;
    UA_Server_addDataSourceVariableNode(m_server,
        m_nodeid,
        m_pnodeid,
        m_referenceTypeId,
        UA_QUALIFIEDNAME(1, qname.data()), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
        vAttr, paramDataSource,
        nullptr, nullptr);
    UA_Server_addReference(m_server,
        m_nodeid,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
        UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY), true);`

Used CMake options:
-DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_HISTORIZING=ON

The node works as expected.  I don't remember seeing this message in 1.1.2.  Want to verify I don't have larger problems related to my nodes.

## Checklist
Please provide the following information:

 - [ x] open62541 Version (release number or git tag): 1.2
 - [x ] Operating system: Ubuntu 18.04
 - [ x] Self-contained code example attached: code from larger system, should be boiler plate default variable setup from examples
 - [ x] Critical issue: non-critical, misunderstanding of why it is giving me this message.  I read the source code of where the message occurs and I am still not understanding the issue.
 - relevant source code location: ua_services_nodemanagement.c line: 322

JensMertelmeyer, 2021-08-09:
To my surprise, `readItemValue` (your ` paramDataSource.read`) will be read by calling `UA_Server_addDataSourceVariableNode(..)`. I suppose your `readItemValue` does not return a value at this point? If your dataSource would not have a writeHandler, the whole  `UA_Server_addDataSourceVariableNode(..)` operation would fail, never adding the node at all.

SchutzSeb, 2022-08-09:
Hi,

I have also meet the same type of messages while running the DI nodeset on my server:

```
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15031): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15032): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15033): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15036): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15037): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15038): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15039): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15040): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15041): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15042): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15043): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15044): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15045): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15046): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15047): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15049): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15050): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15086): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15087): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15088): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15089): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15090): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15091): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15092): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15093): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15094): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15095): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15096): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15097): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15098): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[... cut ...]
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=6534): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[info/session]: SecureChannel 0 | Session "Administrator" | AddNode (ns=2;i=15052): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
```

And I don't understand how to fix this, except setting allowEmptyVariables of the server config to UA_RULEHANDLING_ACCEPT, but that's just avoiding the issue.

What is it recommended to do in this situation ?

piperoc, 2022-09-03:
I'm still seeing this message and I am not sure I understand it, especially the "Create a matching default value" suggestion. when the variable is created I set an initial value like this:

```cpp
varAttr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
UA_Int32 varInt = 0;
UA_Variant_setScalar(&varAttr.value, &varInt, &UA_TYPES[UA_TYPES_INT32]);  

```
is there any other way to create a default?