open62541/open62541 #4318: Single callback for all data change notifications.

andigor, 2021-04-08:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [master] open62541 Version (release number or git tag):
 - [none] Other OPC UA SDKs used (client or server): 
 - [windows] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Hello,

Thanks for reading this.

Background.
I am building an OPCUA gateway to another protocol. I am using ``UA_Client_MonitoredItems_createDataChange``  to subscribe and monitor data changes. As soon as notification arrives I am bypassing data further through the gateway.

The problem.
Data change notifications are delivered one by one. That means I also need to send the data one by one through my gateway which is quite a heavy operation in my case. It would be much simpler if I could receive whole bunch of data in one call.

I do see that ``dataChangeCallback`` is called in the loop from ``processDataChangeNotification`` and delivers every changed data item one by one. I do see that ``processDataChangeNotification`` receives all the data at once via ``UA_DataChangeNotification``. This is exactly what I need. However I don't see a way how I can access ``UA_DataChangeNotification`` from my code.

Question.
Basing on the description above - is there a way to receive all active data change notifications in a single callback call? (same as ``processDataChangeNotification``) ?

Regards,
Andrii

andigor, 2021-04-08:
it could also help if there was a way to determine the amount of changed data items. I could the organize a counter in the callback and act as soon as the callback is called last time.

andigor, 2021-04-09:
Another thing is that the call itself looks like an exponential complexity:
```
static void
processDataChangeNotification(UA_Client *client, UA_Client_Subscription *sub,
                              UA_DataChangeNotification *dataChangeNotification) {
    for(size_t j = 0; j < dataChangeNotification->monitoredItemsSize; ++j) {
        UA_MonitoredItemNotification *min = &dataChangeNotification->monitoredItems[j];

        /* Find the MonitoredItem */
        UA_Client_MonitoredItem *mon;
        LIST_FOREACH(mon, &sub->monitoredItems, listEntry) {
            if(mon->clientHandle == min->clientHandle)
                break;
        
}
```

first loop goes via ``monitoredItemsSize`` and then for each its iteration there is an internal ``LIST_FOREACH`` loop  via all the ``monitoredItems``. So it feels like it can be slow if the amount of ``monitoredItems`` is big and many items change at the same time. 

So it feels like providing a possibility for to have a custom callback instead of  default ``processDataChangeNotification`` is a good idea anyway. In this case users can deal items searching him themselves. For instance using a c++ ``unordered_map`` could have been an option

xydan83, 2022-04-04:
I have same trouble. 
If I try to subscribe to 30,000 nodes that change at the same time once a second, then this filtering algorithm causes a very heavy load on the processor core. There is a quadratic complexity O(n ^ 2). It is quite possible to remove this problem by replacing the second loop with a lookup in a dictionary (hash table).
And it would also be cool if you could make a callback function that would return all the changes at once, in which I agree with the author of this issue.