open62541/open62541 #4326: CMake says "open62541" is considered to be NOT FOUND."

MartinMajewski, 2021-04-09:
## Description

I just followed the installation tutorial `https://open62541.org/doc/current/installing.html#manual-installation`.
1. git clone of the master branch
2. git submodule update --init --recursive
3. mkdir build && cd build
4. cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
5. make
6. sudo make install

Everything went smoothly from what I can tell.

I started a new CMake project with this CMakeList.txt file:

```
cmake_minimum_required(VERSION 3.17)
set(APP_NAME "upc_ua")

project(${APP_NAME} VERSION 1.0.0 LANGUAGES C)

set(CMAKE_C_STANDARD 99)

find_package(open62541 REQUIRED COMPONENTS Events FullNamespace)
add_executable(${APP_NAME} main.cpp)
target_link_libraries(${APP_NAME} open62541::open62541)

add_executable(${APP_NAME} main.c)
target_link_libraries(${APP_NAME} open62541::open62541)
```

The following error occurred:

```
/usr/local/bin/cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_COMPILER=/usr/bin/gcc-8 -DCMAKE_CXX_COMPILER=/usr/bin/g++-8 -G "CodeBlocks - Unix Makefiles" /home/pi/dev/projects/opc_ua
CMake Warning at /usr/local/lib/cmake/open62541/open62541Config.cmake:53 (message):
  Unsupported open62541 component: Events
Call Stack (most recent call first):
  CMakeLists.txt:8 (find_package)

CMake Error at CMakeLists.txt:8 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/open62541/open62541Config.cmake

  but it set open62541_FOUND to FALSE so package "open62541" is considered to
  be NOT FOUND.
```

When I remove "Events" from the find_package call, CMake configures "successfully".

"successfully" in quotes, because I cannot include the header files into my main.c file, as they are not being found.

What is going on here? Can anyone please assist me?

Best wishes,
Martin

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.2-212-g994dff95
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Rasberry OS 64Bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

ccvca, 2021-04-09:
>  Unsupported open62541 component: Events
You did not build with event support.

MartinMajewski, 2021-04-10:
Okay, I see, thanks.

However, this is very confusing for a newcomer, as building, installing and the CMake part is just a few lines apart in the documentation. As this is the starting point it should be coherent and not misleading. My proposal is to either include the Events component in the installing steps, or remove it from the CMake section, or at least give a hint that it will not work out of the box as is. However, the documentation states:

"In your own CMake project you can **then** include the open62541 library using:"

"Then" implies that it is a direct conclusion to the previously provided information, which it isn't.

Furthermore, with this snippet, the open62541 headers are not available for the following minimal server example. As I am no CMake guru I used the approach to build the server with -DUA_ENABLE_AMALGAMATION=ON to get the standalone all-included header file which I have then copied to my project folder. As the documentation states that Amalgamation is not preferred, it should be pointed out how to use the system-wide installed headers.

Don't get me wrong, all I'm saying is to make the start easier when firstly dealing with open62541.

Cheers.

Phil0707, 2023-01-16:
> You did not build with event support.

How can i build with event support?