open62541/open62541 #4327: open62541 set up custom server

Psoke, 2021-04-10:
## Description
Hey there I have some basic questions about setting up information model on my open62541 server. My task is to take a given information model (custom nodes and a small state machine) from a paper and implement it on my own server. I have know about the [OPC UA Rocks tutorial](https://opcua.rocks/from-modelling-to-execution-opc-ua-information-model-tutorial/) but I do not quite understand parts of it.

I took the open62541 code from github and built it on my Windows10 PC with the help of cmake and Visual Studio 16 2019. Till now I do not understand all the options I used (shared libs, amalgamation) but I have a build directory with the needed open62541.h/.c.
Now I started a new Visual Studio project with these two files and a server.c containing the simple example code from the open62541 documentation, built it and voila, a nice little Server.exe that is working and can be accessed from any other PC in my network.
But now I do not know how to proceed. I would like to setup and later modify the information model using a GUI, rather than coding it in Visual Studio. So I tried programs like UAModeler, UMX Pro and such. But the options given do confuse me. For example in UAModeler. Do I need an ansi c, cpp or net project? Or just the modelling option? What files do I need? Every tutorial referes to a "Model.xml" and corresponding .csv file. Then I should use for example the [UA Model Compiler](https://github.com/OPCFoundation/UA-ModelCompiler) to convert the files to a NodeSet2.xml file. But what next?

I thought you need a .c/.h file that bolds the information model I designed in any of the GUI programs to include it in the Visual Studio project, modify the server.c so they will be included in the build process and just build a new Server.exe?

I tried some example .xml and .csv files from the [animal example server](https://github.com/Pro/opcua-animal-cs) and tried to compile them with the UA Model Compiler after successfully building it with Visual Studio, but it keeps giving me error messages about some "no identifier file specified", something wrong with a Programm.cs file.

I would like to discuss my error messages in detail but first I need a kind soul that would show me the way of achieving my goal and explaining the best strategy and order of programs to use.

thank you

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.2 (current)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

johndzilva, 2021-04-12:
To give you a start, you may need to understand the information modeling pipeline [https://opcua.rocks/wp-content/uploads/2019/05/Pipeline.png](url). As you have already encountered, there are two types of xml files. Model.xml is considered as a simple hand written xml file which only contains the user defined nodes. Other one is Nodeset2.xml file. Which is, according to the specification, considered as an interoperable meta file for OPC UA. 

Now coming to your question,

> For example in UAModeler. Do I need an ansi c, cpp or net project? Or just the modelling option? What files do I need? Every tutorial referes to a "Model.xml" and corresponding .csv file. Then I should use for example the UA Model Compiler to convert the files to a NodeSet2.xml file. But what next?

Typically the licensed model compilers directly give you the .exe or binary file of server after compilation. But if you want to use open62541 stack for generating the server, then you should just export the NodeSet2.xml file from it. And use it with the nodeset compiler `https://github.com/open62541/open62541/blob/master/tools/nodeset_compiler/nodeset_compiler.py` to generate .c/.h files which in turn will be used to compile your custom information modeling OPC UA server.

Check the compilation of this example to know about the nodeset compiler `https://github.com/open62541/open62541/tree/master/examples/nodeset`

EvgKlin, 2021-04-13:
@johndzilva NodeSet2.xml file and xml for my nodes i should write by myself, by hand. Or can I use some graphical tools?

johndzilva, 2021-04-13:
@EvgKlin NodeSet2.xml file is the standard schema file. 
Use some modeling tool. Generate the output file as xml. After that use nodeset compiler to generate .c/.h files.

EvgKlin, 2021-04-13:
@johndzilva Which tool you would recommend

Psoke, 2021-04-15:
Yes which tool is recommended? 
UaModeler only supports 10 Nodes --> useless
Beeond UMX Pro only supports 20 Nodes --> less useless but still useless
FreeOPCUA Modeler --> I dont get it to work on Windows, installed it have a .exe, but it closes everytime and I dont get the GUI shown in the github Documentation

Is there no easy free GUI out there for setting up custom Information Models, do I really need to code the nodes by hand?

mlang-de, 2021-04-15:
Maybe the Siemens Siome is an option.
https://support.industry.siemens.com/cs/document/109755133/siemens-opc-ua-modeling-editor-(siome)-zur-umsetzung-von-opc-ua-companion-spezifikationen?dti=0&lc=de-WW

EvgKlin, 2021-04-15:
@Psoke, i get problems wth installation with FreeOPCUA Modeler too.

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

Model your nodes in an editor that exports NodeSet2 XML, generate C code with the open62541 NodeSet compiler (or load the XML at runtime) and build it into your server.

Closing for now.

closed 2026-10-20 by o6-nbeier