open62541/open62541 #4328: Problem on using nodeset-compiler with ISA-95 nodeset

hanyangyexun, 2021-04-12:
## Description
Hello,
I was trying to use the nodeset-compiler function to compile the ISA-95 nodeseet (from OPC foundation nodeset).

According to the open62541 manual, every step was successful but there was error when executing the final OPC UA server (-o file).

I also tried the same process for compile the IEC 61850-7-3 and the MDIS nodeset, they are successful.

This is strange and it is hard for me to locate the problem.

Did anyone have experience on nodeset-compiler?

My guess was that the ISA-95 nodeset file was outdated as it was developed a long time ago in 2013, the open62541 cannot parse it now.

Thank you.
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): UA Modeler 1.65
 - [ ] Operating system: Ubuntu 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: error/server
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-04-14:
> My guess was that the ISA-95 nodeset file was outdated as it was developed a long time ago in 2013, the open62541 cannot parse it now.

If it compiles it should also build.

Can you provide more information than just "error when executing the final OPC UA server (-o file)."

Please either provide log output, a valgrind trace or the generated code file so we can reproduce with minimum effort.

hanyangyexun, 2021-04-14:
Thank you for your reply.

To reproduce, the following are my development processes by using the nodeset_compiler from open62541:

1. generate ISA-95 namespace (`ua_namespace_isa95.c/.h` files) from `Opc.ISA95.NodeSet2.xml` file.

[image]

2. generate user-specific namespace (`isa95atest.c/.h` files) from `isa95atest.xml` exported from UA Modeler.

[image]

3. make a main server `isa95_server.c` and compile all the c files.

[image]

4. run `isa95_server`

[image]

All used files can be accessed at [here](https://github.com/hanyangyexun/OPC-UA-for-ISA-95).

Thank you again.