open62541/open62541 #4334: PubSub MQTT - subscriber crashed after 256 cycles

jackybek, 2021-04-13:
## Description
Still get the following core dump when subscribing from mqtt broker:

myNewServer: malloc.c:2385: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

## Background Information / Reproduction Steps
OPCUA Server1 Publish to MQTT broker - no issue
OPCUA Server2 subscribe from MQTT broker - core dump after running for a few cycles

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.2.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: raspbian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-04-13:
Stack trace?

jackybek, 2021-04-14:
Hi Julius, I did a trace and realised the cause of this crash is due to the library from json.org/json_checker .. since this is free library, I guess it was not built for production environment.  I also read somewhere in this forum that open62541 may introduce its own json checking algorithm.  I will wait for this version.

closed 2021-04-14 by jackybek