open62541/open62541 #4336: PubSub over MQTT: Running tutorial_pubsub_mqtt_publish failure!

DanielZYH, 2021-04-14:
## Description
I run the example code without modification for tutorial_pubsub_mqtt_publish.c on Ubuntu18.04, the application print "register channel error". And then I checked the mosquitto(mqtt broker), the broker printed out "Socker error on client xxxxxxxx". I tried to use another mqtt client tools to connect this mqtt broker, the result is connected. So I suppose there is an error about mqtt client for tutorial_pubsub_mqtt_publish.c, could you give me help?

## Background Information / Reproduction Steps
Step1:  start mqtt broker
daniel@ubuntu:~$ sudo mosquitto
[sudo] password for daniel: 
1618370997: mosquitto version 1.5.5 starting
1618370997: Using default config.
1618370997: Opening ipv4 listen socket on port 1883.
1618370997: Opening ipv6 listen socket on port 1883.

Step2: run pubsub mqtt instance.
root@ubuntu:/home/daniel/work/opcua/open62541/build/bin/examples# ./tutorial_pubsub_mqtt_publish 
[2021-04-07 02:29:47.671 (UTC-0700)] info/session	SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNode (i=15303): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[2021-04-07 02:29:47.688 (UTC-0700)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-04-07 02:29:47.694 (UTC-0700)] info/server	AccessControl: Anonymous login is enabled
[2021-04-07 02:29:47.694 (UTC-0700)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2021-04-07 02:29:47.694 (UTC-0700)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-04-07 02:29:47.695 (UTC-0700)] info/userland	PubSub MQTT: channel requested.
[2021-04-07 02:29:47.805 (UTC-0700)] info/server	PubSub MQTT Connection established.
[2021-04-07 02:29:47.806 (UTC-0700)] info/server	MetaData creation. Found DataType UtcTime.
[2021-04-07 02:29:47.806 (UTC-0700)] warn/server	register channel failed: 0x80760000!
[2021-04-07 02:29:47.807 (UTC-0700)] error/server	Error Name = BadArgumentsMissing

step3: check mosquitto logs
daniel@ubuntu:~$ sudo mosquitto
[sudo] password for daniel: 
1618370997: mosquitto version 1.5.5 starting
1618370997: Using default config.
1618370997: Opening ipv4 listen socket on port 1883.
1618370997: Opening ipv6 listen socket on port 1883.
1618371008: New connection from 127.0.0.1 on port 1883.
1618371009: New client connected from 127.0.0.1 as TESTCLIENTPUBSUBMQTT (c0, k400).
1618371009: Socket error on client TESTCLIENTPUBSUBMQTT, disconnecting.
1618371441: New connection from 192.168.3.120 on port 1883.
1618371441: New client connected from 192.168.3.120 as d48cb879-bb96-404a-95b9-bc9f7d7fbb941618371441724 (c1, k10).
**1618371501: Socket error on client d48cb879-bb96-404a-95b9-bc9f7d7fbb941618371441724, disconnecting.**
1618371501: New connection from 192.168.3.120 on port 1883.
1618371501: New client connected from 192.168.3.120 as d48cb879-bb96-404a-95b9-bc9f7d7fbb941618371502058 (c1, k500).
1618371504: Socket error on client d48cb879-bb96-404a-95b9-bc9f7d7fbb941618371502058, disconnecting.
1618371506: New connection from 192.168.3.120 on port 1883.
1618371506: New client connected from 192.168.3.120 as d48cb879-bb96-404a-95b9-bc9f7d7fbb941618371507526 (c1, k500).
1618375946: Socket error on client d48cb879-bb96-404a-95b9-bc9f7d7fbb941618371507526, disconnecting.
1618375946: New connection from 192.168.3.120 on port 1883.
1618375946: New client connected from 192.168.3.120 as d48cb879-bb96-404a-95b9-bc9f7d7fbb941618375947238 (c1, k500).
1618375948: Socket error on client d48cb879-bb96-404a-95b9-bc9f7d7fbb941618375947238, disconnecting.

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON -DUA_ENABLE_PUBSUB_MQTT=ON ..
```

## Checklist

Please provide the following information:

 - [#] open62541 Version (release number or git tag):  Master branch
 - [#] Other OPC UA SDKs used (client or server): server
 - [#] Operating system:  Ubuntu 18.04 , mqtt broker(mosquitto 1.5.5)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2021-04-29:
There are additional parameters that needs to be initialised before the connection to mqtt can be successful.  You can look for my reply to an earlier question which is similar nature.

shaohuazhang1, 2022-09-26:
sudo mosquitto
1664159347: mosquitto version 2.0.15 starting
1664159347: Using default config.
1664159347: Starting in local only mode. Connections will only be possible from clients running on this machine.
1664159347: Create a configuration file which defines a listener to allow remote access.
1664159347: For more details see https://mosquitto.org/documentation/authentication-methods/
1664159347: Opening ipv4 listen socket on port 1883.
1664159347: Error: Address already in use
1664159347: Opening ipv6 listen socket on port 1883.
1664159347: Error: Address already in use
I am not operating any application on port 1883,why error occur?

shaohuazhang1, 2022-09-26:
I found mosquitto running automatively after installed.So the port is in us.

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

The invalid registration call in the publisher was removed in #4235, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier