open62541/open62541 #4337: server_nodeset_powerlink failled

EvgKlin, 2021-04-14:
## Description
Example server_nodeset_powerlink failled at  `UA_StatusCode retval = namespace_di_generated(server);` 
server_nodeset_powerlink is not working too. 
server_nodeset works well
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-04-14:
Right.
It seems the generated value-variant contains an array but does not have ArrayDimensions.
But the ArrayDimensions are expected to be present in the value if they are defined for the Variable. We will fix this.

EvgKlin, 2021-04-14:
Is there other possibility to see nodes and information model? or can i export this xml to UaModeler

jpfr, 2021-04-15:
You can directly import the XML in UAModeler.
Its a standard format.

EvgKlin, 2021-04-15:
Small question, view nodes are not implemented or?

closed 2021-08-30 by EvgKlin