open62541/open62541 #4340: Compatibility of open62541 library with PLC's of SIMATIC controllers, Allen Bradley PLCs and Programmable Controllers MELSEC

Kishan-Parmar1425, 2021-04-16:
I am using OPC-UA client from example.

Does my OPC-UA client can connect to PLC's like  SIMATIC controllers, Allen Bradley PLCs and Programmable Controllers MELSEC?

Or is there any dependency to make a connection or to read parameters from those PLC's ?

jpfr, 2021-04-16:
Yes, the client can connect.
This is tested regulalrly against common OPC UA implementations.

If you cannot connect to a particular server, we will treat it as a bug on our end and fix it.
Even if the other server is not 100% compliant. To ensure we are compatible with devices in the field.

closed 2021-04-16 by jpfr

Kishan-Parmar1425, 2021-04-16:
Thank you for your quick response.

What should be done in case of some old PLC's on which OPC server is not running, can we make connection to them without using any third party tools?

Any input on it.