open62541/open62541 #4342: Can i implement OPC UA applikation like Server and like Client?

EvgKlin, 2021-04-16:
## Description
Can i implement OPC UA applikation like Server and like Client? Because i see some examples im Internet and it should be possible (see pic). 

## Background Information / Reproduction Steps[image]

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2021-04-17:
Yes .. you can create 1 separate thread in your main() program : 
1. The main program will launch UA server
2. The main program spawns a new thread and within this thread you can launch a UA client.

EvgKlin, 2021-04-20:
@jackybek Thank you. Can i ask one question: I can add nodes to server from the client. But can i add nodes from server to client? Or is it realized only over subscribing and request/response

jackybek, 2021-04-20:
A opcua client does not have any nodes.  Only a opcua server has the nodes. So you can either:
1. Within a server code, add nodes to the address space
2. 2. Within a client code, send a request to the server to add new nodes

EvgKlin, 2021-04-21:
@jackybek sorry for interrupting, there is other issue(https://github.com/open62541/open62541/issues/4353#issue-862735830), but i get the same problem. if use amalgamation in Windows, i get errors. In Debian it works perfectly. Maybe do i something wrong

jackybek, 2021-04-30:
Sorry .. I am using Linux version and has not played with the windows version of open62541.

andreasebner, 2025-07-30:
The original question seems to be answered. Amalgamation build are tested as part of the CI, so they should work.